pantor/ruckig #5: Unexpected behavior of `output` when `update` returns `Result::Done`

isherman, 2021-03-01:
Thanks for the great library, I've been enjoying working with it.

I got bit today by some behavior that may be intended, but was surprising to me. I had a loop that looked something like the following:

```
while (...) {
  if (has_new_setpoint) {
    input.target_position = new_setpoint;
  }
  auto result =  otg.update(input, output);
  if(result == Result::Working || result == result::Done) {
    input.current_position = output.new_position;
    //...
  }  
}
```
I eventually realized that `output.new_position` is uninitialized or in an undefined state when `result == result::Done`. Perhaps this behavior could be documented, or enforced with a runtime error. Or perhaps `output`'s fields should hold meaningful values when `result == result::Done`.

pantor, 2021-03-02:
Hi @isherman! Uninitialized output values are definitely a bug - Ruckig should output valid parameters after the trajectory has finished with `Result::Finished`. By default, Ruckig continues the trajectory with zero jerk.

However I can't replicate the issue. Can you give me a self-contained code example? Are you sure you are using the latest version of Ruckig?

isherman, 2021-03-02:
Sure. Apologies this isn't a proper automated test, but here's the modified `examples/position.cpp` that results in the unexpected behavior for me (at HEAD of `master`: e9af3211e2360cf4dc49cf9d86d4ba728181c272)

```
#include <iostream>

#include <ruckig/parameter.hpp>
#include <ruckig/ruckig.hpp>

using namespace ruckig;

int main() {
    // Create instances: the ruckig otg as well as input and output parameters
    Ruckig<6> otg {0.001};
    InputParameter<6> input;
    OutputParameter<6> output;

    // Set input parameters
    input.max_velocity = {1.2, 1.2, 1.2, 0.6, 0.6, 0.6};
    input.max_acceleration = {4.0, 4.0, 4.0, 1.5, 1.5, 1.5};
    input.max_jerk = {10.0, 10.0, 10.0, 4.0, 4.0, 4.0};

    input.current_position = {0.0, -M_PI / 2.0, M_PI / 2.0, 0.0, 0.0, 0.0};
    input.current_velocity = {0.0, 0.0, 0.0, 0.0, 0.0, 0.0};
    input.current_acceleration = {0.0, 0.0, 0.0, 0.0, 0.0, 0.0};

    input.target_position = {0.0, -M_PI / 2.0, M_PI / 2.0, 0.0, 0.0, 0.0};
    input.target_velocity = {0.0, 0.0, 0.0, 0.0, 0.0, 0.0};
    input.target_acceleration = {0.0, 0.0, 0.0, 0.0, 0.0, 0.0};

    // Generate trajectory
    std::cout << "t | p1 | p2 | p3 | p4 | p5 | p6" << std::endl;
    auto result = otg.update(input, output);
    do
    {
        auto &p = output.new_position;
        std::cout << output.time << " " << p[0] << " " << p[1] << " " << p[2] << " " << p[3] << " "
                  << p[4] << " " << p[5] << std::endl;

        input.current_position = output.new_position;
        input.current_velocity = output.new_velocity;
        input.current_acceleration = output.new_acceleration;
    } while (result == Result::Working);

    std::cout
        << "Reached target position in " << output.trajectory.duration << " [s]."
        << std::endl;
}
```

pantor, 2021-03-02:
There was indeed a bug for trajectory with zero duration. The example should be fixed with the latest commit in master.

isherman, 2021-03-02:
Confirmed. Thank you @pantor! Closing.

closed 2021-03-02 by isherman