pantor/ruckig #6: Python Example

ntorresalberto, 2021-03-03:
First of all, thank you very much for this work!

Second, I'm very interested in trying it on python as it would help prototyping but I'm not sure how to use it.
I've run `cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_MODULE=True ..` and it seems building went well:

```
→ cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_MODULE=True ..
-- Found PythonInterp: /usr/bin/python (found version "3.8.7") 
-- Found PythonLibs: /usr/lib64/libpython3.8.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /usr/include (found version "2.6.2" )
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/src/github/ruckig/build

→ make
[  8%] Building CXX object CMakeFiles/ruckig.dir/src/brake.cpp.o
[ 16%] Building CXX object CMakeFiles/ruckig.dir/src/position1.cpp.o
[ 25%] Building CXX object CMakeFiles/ruckig.dir/src/position2.cpp.o
[ 33%] Building CXX object CMakeFiles/ruckig.dir/src/velocity1.cpp.o
[ 41%] Building CXX object CMakeFiles/ruckig.dir/src/velocity2.cpp.o
[ 50%] Linking CXX shared library libruckig.so
[ 50%] Built target ruckig
[ 58%] Building CXX object CMakeFiles/otg-test.dir/test/otg-test.cpp.o
[ 66%] Linking CXX executable otg-test
[ 66%] Built target otg-test
[ 75%] Building CXX object CMakeFiles/_ruckig.dir/src/python.cpp.o
[ 83%] Linking CXX shared module _ruckig.cpython-38-x86_64-linux-gnu.so
[ 83%] Built target _ruckig
[ 91%] Building CXX object CMakeFiles/example-position.dir/examples/position.cpp.o
[100%] Linking CXX executable example-position
[100%] Built target example-position
```

Could you provide a minimal working example of how I would import and call the library?
I'm ok to work out the details with respect to the C++ API.

pantor, 2021-03-03:
Hi! You can have a look at [this](https://github.com/pantor/ruckig/blob/master/test/otg_plot.py) python script for plotting trajectories.

In summary, you have to import the module named `_ruckig`. In order to find the module, you have to import the build directory into your `PYTHONPATH` variable, e.g. by `sys.path.insert(0, '<path to ruckig>/build'))`.

ntorresalberto, 2021-03-03:
The example you mention seems to throw an error:
```
Traceback (most recent call last):
  File "example_script2.py", line 11, in <module>
    from _ruckig import Reflexxes
ImportError: cannot import name 'Reflexxes' from '_ruckig' (/home/user/src/github/ruckig/build/_ruckig.cpython-38-x86_64-linux-gnu.so)
```

I'm running it on the `build/` directory, where the `_ruckig.cpython-38-x86_64-linux-gnu.so` is located.

On the other hand, other stuff is working as I was just doing a minimal example that seems to work, I'll leave it here in case it's useful to someone else:

(run inside `build/` directory or add it to the python path)
```Python
import _ruckig as ruc
import matplotlib.pyplot as plt

dt = 1e-3
ruckig = ruc.Ruckig(dt)
inp = ruc.InputParameter()
inp.max_velocity = 3 * [.5]
inp.max_acceleration = 3 * [2]
inp.max_jerk = 3 * [10]
inp.target_position = 3 * [2]

out = ruc.OutputParameter()

t = [0]
x = [inp.current_position[0]]
xd = [inp.current_velocity[0]]
xdd = [inp.current_acceleration[0]]

result = ruckig.update(inp, out)
while result == ruc.Working:
    inp.current_position = out.new_position
    inp.current_velocity = out.new_velocity
    inp.current_acceleration = out.new_acceleration

    t.append(t[-1] + dt)
    x.append(out.new_position[0])
    xd.append(out.new_velocity[0])
    xdd.append(out.new_acceleration[0])
    result = ruckig.update(inp, out)

plt.plot(t, x, label='x')
plt.plot(t, xd, label='xd')
plt.plot(t, xdd, label='xdd')

plt.legend(shadow=True,
           fancybox=True,
           loc="upper left",
           bbox_to_anchor=(1, 1.05))

plt.ylabel('[m]')
plt.xlabel('[s]')

plt.show()
```

pantor, 2021-03-03:
Great that you got it working.

Reflexxes is only available in the Python module if Ruckig was compiled with the Reflexxes option.

The number of DoFs is a compile-time parameter in C++, so you can't change it in Python afterwards. This was primarily done for performance reasons. It is indeed not a perfect solution for Python, so I'm thinking about adding some typical numbers (like 1, 3, 6, 7) for DoFs manually.

ntorresalberto, 2021-03-03:
> Reflexxes is only available in the Python module if Ruckig was compiled with the Reflexxes option.

From the readme it seemed to only be used for comparisons, am I missing somthing? are some features only available when compiled with it?

> The number of DoFs is a compile-time parameter in C++, so you can't change it in Python afterwards. This was primarily done for performance reasons. It is indeed not a perfect solution for Python, so I'm thinking about adding some typical numbers (like 1, 3, 6, 7) for DoFs manually.

Yeah, I saw. And I greatly appreciate you guys making optimal code. It looks really well thought out.
If I may, what you propose seems similar to what's done with the popular [Eigen library](https://eigen.tuxfamily.org/dox/group__TutorialMatrixClass.html) (with the  `Vector2d`, `Vector3d`, etc classes) so it seems like a very intuitive choice.

pantor, 2021-03-03:
Yeah, Reflexxes is only in the Python module (and in the test script linked above) for comparison purposes. It is not needed for any features of Ruckig.

pantor, 2021-03-03:
I've added a Python example in the `example/` folder, so I'll close this issue.

closed 2021-03-03 by pantor