pantor/ruckig #7: Feature: Constrained trajectory or additional

peetCreative, 2021-03-11:
Hey,
I'm currently working on an application, in which  the robot-arm moves around a point, always pointing towards it.
So I need to constraint the trajectory in every step. For me the constraint would be that the orientation of the end-effector is pointing towards this certain point. Do you think it's possible to add like a constraint function to the input parameters?

Another more relaxed idea is to define several intermediate target poses in the input parameters.
I don't want the trajectory to stop at these point just to fly by.

What do you think would be the complexity to add these features? Which parts should I look at?
However I really want to thank you for your work.
Best regards,
Peter

pantor, 2021-03-11:
Hi, thanks for your input!

1. I think that this is out of scope for Ruckig as a general OTG algorithm. The general solution would be to parametrize your system regarding your constraints with fewer DoF. So in your case, just use Ruckig with 3 DoF for the translation, and calculate the orientation accordingly.

2. A general path-based OTG, so real-time time-parametrization with jerk constraints, is a hard task. In the long-term, I would like to integrate that into Ruckig (see the [path](https://github.com/pantor/ruckig/tree/path) branch for first ideas), but I can't promise anything as time is limited...

closed 2021-03-17 by pantor

pantor, 2021-09-09:
Hi @peetCreative and @gavanderhoorn,

just for your information: We now offer a closed-source version of Ruckig that supports intermediate waypoints. In comparison to offline approaches like TOTP or TOPP-RA, Ruckig offers: (1) faster trajectory durations due to joint path and time parametrization, (2) real-time capability, and (3) jerk limits. You can find more information at [ruckig.com](https://ruckig.com), and also some examples and docs in the Readme here. More details and comparisons are coming soon!

gavanderhoorn, 2021-09-09:
Thanks for the update. :+1: 

I'm happy for you, but sad for the OSS side of things.

Don't get me wrong: selling software is completely OK (and even if I thought it wasn't, that doesn't matter at all). If I'm ever in a situation where I have budget for acquiring a license I'll keep it in mind.

pantor, 2021-09-10:
Yes, I feel you very well. I've looked into a few approaches for selling Ruckig as OSS, but I fear that they are not really feasible (at least for now). Hopefully, I can release more-and-more features as open source when I'm starting to integrate new features into the closed-source version.