pantor/ruckig #13: Feature : Phase synchronization trajectory.

Linjackffy, 2021-03-22:
Hi, all.
Is it impossible to integrate phase synchronization trajectory, which is supported by Reflexxes, into ruckig?

pantor, 2021-03-25:
Hi @Linjackffy,

this is certainly possible and shouldn't be that difficult. However, I don't like the phase synchronization feature from an API design perspective. I think it would be better to implement a real path-following OTG, even if it would be limited to linear paths at first. Nonetheless, I would be happy to accept PRs for phase synchronization!

jonbinney, 2021-03-25:
I'm not familiar with reflexxes - is phase synchronization intended to make sure that the joints stay exactly on the path in joint space between start and goal?

pantor, 2021-03-25:
Phase synchronization makes sure that the path of the trajectory is *linear*, when possible (so when all derivatives are collinear).

jonbinney, 2021-03-25:
Makes sense - when only a start and a goal are given, as @pantor mentioned, this does seem like a specific case of path following for a linear path. I've solved this problem before by first parameterizing the path with respect to a single variable, `f(s)`, where `f(s)` is the position of the robot joints at a distance `s` along the path. "distance" can be a bit arbitrary, but it is convenient to use the cumulative arc length in joint space.

Once the path has been described as `f(s)`, time parameterization is only a single DOF problem - finding `s = g(t)`. The tricky part is that we have to convert the limits on the joint positions/velocities/accelerations into limits on the single variable `s`. For a linear path segment, I believe that is as simple as projecting the limits onto the direction vector of the path segment, and then choosing the most restrictive values (taking the intersection of the allowable values of s according to each joint's limits). For curved path segments, projecting the limits onto `s` is more tricky, but I think there are probably reasonable approaches.

I've implemented that approach in python here: https://github.com/jonbinney/trajectory_smoothing/blob/0e2b8d7d646c96c0c22eef1371bcd42d169121dc/traj/src/traj/trajectory.py#L10

That worked well for simple  trajectories with starting and ending jerks and accelerations of zero. I've been working on a type 5 trajectory version following a similar approach to [1], but implementing the decision tree and solving all the cases is quite a challenging project. Which is why I'm happy to see Ruckig :smile: 

[1] T. Kröger and F. M. Wahl, "Online Trajectory Generation: Basic Concepts for Instantaneous Reactions to Unforeseen Events," in IEEE Transactions on Robotics, vol. 26, no. 1, pp. 94-111, Feb. 2010, doi: 10.1109/TRO.2009.2035744.

pantor, 2021-03-25:
I would really like to integrate path-following time parametrization into Ruckig. But I'm afraid that I won't have time for that in the near future... Thanks for sharing your approach!

Btw, Reflexxes is written by T. Kröger (who happens to be my supervisor), so it is basically the approach from [1]. And Ruckig works without decision trees, that's the cool thing about it :wink:

jonbinney, 2021-03-25:
@pantor you've already implemented an awesome amount of functionality in ruckig; no need to implement path following. The method I'm suggesting is something that I can implement outside of ruckig, and then use ruckig to solve the remaining 1D problem. I like that ruckig does one thing and does it very well.

It is great to hear that you are working with Torsten and have made such great progress; I'm very impressed with ruckig!

Linjackffy, 2021-03-26:
@pantor,  Parameterizing the path with respect to a single variable `s` is alternative way to make phase synchronization trajectory. For linear path, it is easy to convert  joint's jerk limit to variable `s`.  For curved path, such as bezier or bspline,  it is so difficult since the second and third order of curved path is nonzero, which means that the jerk limit of joint would impact the limit of `s_dot` and  `s_ddot` .  Can ruckig handle variable bound on jerk limit?

jonbinney, 2021-03-29:
@Linjackffy I don't want to speak for @pantor, but since rucking represents the time parameterization using a small number of polynomials, It shouldn't be possible for it to accommodate a variable jerk limit. As an example, if the jerk limit is a sin wave, for example, the optimal time parameterization definitely wouldn't be possible to represent using polynomials.

The fact that the jerk limit depends on `s_dot` and `s_ddot` (instead of just varying with `s`) makes things even more complicated. One thing that i'm interested in trying is to use discrete planning algorithms to try to stitch together an optimal time parameterization using small sections that have been generated by ruckig. Each section could be kept small enough that treating the limits as constant within the section is not too bad.

pantor, 2021-04-04:
I've now decided to implement the phase synchronization setting, as it makes migration from Reflexxes easier. Using the latest commit of the master branch, phase synchronization can be turned on via
```cpp
input.synchronization = Synchronization::Phase;
```
Once this is tested thoroughly, I plan to make it the default synchronization behavior.

pantor, 2021-04-04:
@Linjackffy No, variable jerk limits are not possible.

closed 2021-04-04 by pantor