pantor/ruckig #14: Installation method for python bindings

jonbinney, 2021-03-25:
Currently the bindings get built into a shared library, but that library doesn't end up anywhere that python expects it to be. There are various ways to handle this, each of which has tradeoffs. If there is one that is acceptable to @pantor , I'm happy to make a PR implementing it.

Here are the two options that come to mind:

1. Use pybind_catkin to build the python bindings, and install with cmake's `install` macro. This would require building the package using catkin. That seems like a heavy dependency for non-ROS people, but I suspect it is possible to detect whether the user is in a ROS environment and only use catkin in that case. See example here:https://github.com/arturmiller/pybind11_examples/blob/master/function_call/CMakeLists.txt

2. Add a setup.py and use setuptools to build the bindings. I'm not sure how this will integrate with the cmake process, but there may be a way to hack it to work. Documentation on how to build pybind11 bindings with setuptools: https://pybind11.readthedocs.io/en/stable/compiling.html#building-with-setuptools

There are probably other ways that I don't know about. In general I'm building ruckig with my own code and other 3rd party libraries, and modifying `sys.path` within my scripts seems a bit hacky. I usually use catkin to handle this, but I realize not everyone wants that as a dependency.

/cc @gavanderhoorn

traversaro, 2021-03-25:
A third option is just to use CMake to install the Python bindings without using neither catkin nor setuptools. Unfortunately the possible use cases of people using C++/Python bindings vary, so a rather general approach is https://github.com/traversaro/ruckig/commit/146e7656dbe296d5e32f404cce1e08f4d3450908  (warning: I did not test it explicitly), in which two options can control the installation behavior:
* `RUCKIG_PYTHON_INSTALL_IN_ACTIVE_SITE` if this option is ON, the Python bindings are installed in the active site package directory (so with automatic support for venv or conda environments), similar to what happens when you install with `setup.py` .However, if the active Python interpreter is the system one, this will result in Python packages installed in the `/usr` prefix and and in any case outside the `CMAKE_INSTALL_PREFIX` , so I personally typically prefer to  keep this option by default to OFF.
*  If `RUCKIG_PYTHON_INSTALL_IN_ACTIVE_SITE` is OFF, the installation directory absolute or relative to `CMAKE_INSTALL_PREFIX`  can be specified with the `RUCKIG_PYTHON_INSTALL_DIR`, that is initialized with a reasonable Python-provided value. The user can set this value manually to the desired installation directory.

However, I am not an expert in catkin, so I do not know if this option could cover also the catkin installation workflow.

pantor, 2021-03-25:
From my perspective, Ruckig is primarily a C++ project, as it is designed to be used in the real-time control cycle where Python is typically avoided. So in general I would prefer @traversaro 's suggestion to use CMake for installing the Python library as well.

However, from your (and other feedback) it seems that the Python module is quite useful. That's why I'm thinking about uploading Ruckig to PyPI, so that it can be easily installed via `pip`. Would that help in your case? Do you need *both* the C++ and Python library? Then I would add a `setup.py` file one way or another.

jonbinney, 2021-03-25:
I'd like to contribute patches to ruckig, so I'm looking for a method that works well from source. A pip installable version may be useful for some people, but as you say, many people will need the C++ library so that they can use it in realtime scenarios. My own workflow for this type of thing is:
1. Work out the basic math using sympy and jupyter notebooks
2. Code the algorithm in python using numpy/etc to see if it works
3. Code the algorithm in C++ to make it more useful for robots.

I like @traversaro's suggestion, and I'll do some experimenting with how that would interact with catkin. Ideally I could get catkin to set the `Python3_SITEARCH` cmake variable to the correct location.

pantor, 2021-03-29:
I've now added a `setup.py` file and continuous deployment to PyPI, so you can install the Python module either via
`pip install ruckig` from PyPI or build it yourself via `pip install .` from the current directory. Does that clear up this issue?

jonbinney, 2021-03-29:
Thanks @pantor ! For my own workflow, I'd like to find a good way to have ruckig play nice with catkin. I think I can do that by doing something in cmake to trick setuptools into doing the "right thing" with respect to caktin's devel and install spaces. That is separate from this github issue though, and I consider this issue to be complete.

closed 2021-03-29 by jonbinney