pantor/ruckig #15: Allow various numbers of DOFs using python bindings

jonbinney, 2021-03-25:
Currently the python bindings hardcode the number of DOFs to 3. It is a template parameter in C++, so it does need to be set at compile time. I think it should be feasible to compile for various commonly used DOFs (maybe 1 through 10) and let python select which to use at runtime.

There's some details around how to implement this on the python side - i.e. would there be a dictionary of Ruckig/InputParameter/etc for all of the the implemented DOFs, so the user would use `Ruckig[DOFs]`? Or a factory function that creates a Ruckig instance for the number of DOFs? Or a wrapper for the Ruckig class that takes the number of DOFs as a parameter to the constructor, and internally uses the correct compiled Rucking C++ object.

I'm happy to implement if there is a good way to do this.

pantor, 2021-03-25:
Hi @jonbinney!

I've always had that in mind, so I've implemented it myself now. You can find my approach in the latest commit on the master branch. What do you think about it?

jonbinney, 2021-03-25:
Thanks @pantor, that looks great! I'll try it out and make suggestions if I have any.

closed 2021-03-25 by jonbinney