pantor/ruckig #16: Preprint version of the paper

caomuqing, 2021-03-27:
Hi,

Thanks for this wonderful package. I would like to ask if there is plan to release a preprint version of the academic paper, to popular preprint website like https://arxiv.org ? It would be really helpful for us researchers to understand the technique, and also cite the paper in our research before it is officially published :)

Best Regards,
mq

pantor, 2021-04-04:
Unfortunately, I will not publish a pre-print paper. I hope that a peer-reviewed acceptance is only a matter of weeks :wink:

closed 2021-04-04 by pantor

pantor, 2021-05-12:
The corresponding paper *Jerk-limited Real-time Trajectory Generation with Arbitrary Target States* is now available [here](https://arxiv.org/abs/2105.04830).