pantor/ruckig #20: New target during trajectory generation gets ignored

AndreRigi, 2021-04-06:
Hi,

Right now, if the target is changed when update returns both Result::Working and Result::Finished, it gets ignored. It seems the generator keeps tracking the first target given. 
Is this the expected behavior ? How can I generate a trajectory to a new target when the previous target is reached or close to be reached ?

pantor, 2021-04-06:
Hi @AndreRigi! This seems like a bug, Ruckig should always try to reach the current target of the given input parameters. Do you have a concrete example where your issue occurs?

AndreRigi, 2021-04-06:
This seems to be related to #19, as I also get the return ` -110` error status when I call `update()`  right after a new  `target_position` is set. However, in the subsequent calls of `update()`, I do receive the status  `Result::Working` with the `target_position` remaining the same new set previously. 

Initial inputs are:
```
inputs.max_velocity = {3.0, 3.0, 3.0};
inputs.max_acceleration = {0.5, 0.5, 0.5};
inputs.max_jerk = {0.2, 0.2, 0.2};  

inputs.current_position = {0.0, 0.0, 0.0};
inputs.current_velocity = {0.0, 0.0, 0.0};
inputs.current_acceleration = {0.0, 0.0, 0.0};

inputs.target_position = {-0.301473, -5.47672, 20};
inputs.target_velocity = {0.0, 0.0, 0.0};
inputs.target_acceleration = {0.0, 0.0, 0.0};
```
with cycle rate of 0.1.

At each iteration, the inputs (position, velocity, acceleration) are set to the outputs (new position, ...).

At some point, when the output reaches:

```
outputs.new_position = {-0.275481, -5.34186, 19.865};
outputs.new_velocity = {-0.0203944, -0.251633, 0.254019};
outputs.new_acceleration = {0, 0.275072, -0.318759};
```

The actual position is in the acceptance radius of the previous setpoint and so a new setpoint is sent to `target_position`:
`inputs.target_position = {1.69676, -27.0059, 20}`

At the next call of `update()`, the return is  `-110`.

From there, the return from `update()` is `0` until reaching the previous setpoint. Then the return from `update()` is `1`. The new setpoint seems therefore to be ignored, while `target_position`  is still set to this new value at each iteration of the loop.

pantor, 2021-04-07:
With [this commit](https://github.com/pantor/ruckig/commit/0dd86f85f04006fa937cf95eedd5db8b3b0dd894), Ruckig now recalculates the trajectory when an error happened before. To be precise, it only saves the current input for comparison if the calculation was successful.

This does not fix the underlying issue, but let's discuss this one in #19.

closed 2021-04-07 by pantor

AndreRigi, 2021-04-07:
Thank you for your quick reaction and feedback @pantor. It's indeed working with this commit.