pantor/ruckig #22: Is position overshoot possible?

isherman, 2021-04-27:
Is it possible that the trajectory returned by Ruckig will overshoot the target position `P` of a target state `<p=P, v=0, a=0>`?

My understanding from this [Reflexxes paper](https://cs.stanford.edu/groups/manips/publications/pdfs/Kroeger_2010_TRO.pdf), e.g. section IV.C, is that it _is_ possible, at least in their approach.

If you have a moment to clarify my conceptual understanding, I much appreciate it!

pantor, 2021-04-27:
Overshooting is neither possible with Reflexxes nor with Ruckig. Section IV.C refers to a limiting position interval, which is then however ignored in both Reflexxes and Ruckig. Ruckig explicitly checks for the correctness of its result (within the stated numerical limits), so I'm doubly sure that Ruckig does not overshoot.

isherman, 2021-04-29:
OK, thank you for the clarification!

closed 2021-04-29 by isherman