pantor/ruckig #23: Inverse s->time

ahoarau, 2021-05-07:
Hello,
I have a use case where I need to find the time corresponding to a particular `s` . `s` the position variable.
Right now we have getState(time) --> position, but can we have findTime(position) --> time ?
Thanks !

pantor, 2021-05-12:
Thanks for your suggestion! In general, the inverse is not unique, so we could only have something like `findTime(position) -> [time]` returning a list of times.

ahoarau, 2021-05-15:
The inverse is not unique, so it means `s` is not monotonic ?
I'm guessing in case of overshooting ?
Do we have access to this info in ruckig ? isOvershooting ?

pantor, 2021-05-17:
For the real, spatial position, the inverse is not unique. For example, a trajectory starting from `p=0` with `v>0` with the same target position and the same positive target velocity will cross `p=0` (at least) three times. Nonetheless, I think this features might still be useful.

If you mean a path position parameter (as often found for path-based time parametrization), this inverse is of course unique. However, ruckig does not expose a path position, and an additional interface for that would be required.

For all cases, Ruckig does not overshoot (see #22).

pantor, 2021-07-06:
The latest commit adds the method `get_first_time_at_position`, which returns the time that the trajectory passes a specific position of a given DoF the first time. You can find the documentation [here](https://pantor.github.io/ruckig/classruckig_1_1_trajectory.html#ab7320d79f44f252eb4d0904b56dfa391).

closed 2021-07-06 by pantor

ahoarau, 2021-07-06:
That is awesome ! Thanks @pantor . @traversaro