pantor/ruckig #25: Bug for phase synchronization trajectory.

Linjackffy, 2021-05-25:
I found a bug when using the feature of phase synchronization. The output position of second axis changed from 0 to 200 in one control cycle.

Below is the test case I used:
```
    Ruckig<2> otg{0.004};
    InputParameter<2> input;
    OutputParameter<2> output;

    input.synchronization = Synchronization::Phase;
    input.duration_discretization = DurationDiscretization::Discrete;

    // Set input parameters
    input.max_velocity = { 1380, 1380 };
    input.max_acceleration = { 3588, 3588 };
    input.min_acceleration = { -6900, -6900 };
    input.max_jerk = { 12420, 12420 };

    input.current_position = { 0.0, 0.0 };
    input.current_velocity = { 0.0, 0.0 };
    input.current_acceleration = { 0.0, 0.0 };

    input.target_position = { 400, 200 };
    input.target_velocity = { 0.0, 0.0 };
    input.target_acceleration = { 0.0, 0.0 };
```

[image]

pantor, 2021-05-25:
I'm afraid I can't replicate this issue. On my system, your test case outputs a valid trajectory.

What system / compiler are you using? Can you try to minimize your test case even further? Do you get an error when omitting `min_acceleration` or the `duration_discretization`?

pantor, 2021-05-25:
This should be fixed with the latest commit on master. Thanks for reporting!

Linjackffy, 2021-05-25:
The problem is fixed! Thank you so much!

closed 2021-05-25 by Linjackffy