pantor/ruckig #26: Status of Paths in Ruckig?

gavanderhoorn, 2021-05-27:
Sort-of a follow-up to #7: what's the status of [ruckig@path](/pantor/ruckig/tree/path)?

Not looking for constraint support (as I agree that seems out-of-scope for Ruckig), but processing a path instead of a segment would be very convenient (and the [blend distance](https://github.com/pantor/ruckig/blob/b59e2a2384d560b8e162491a2bb955276df4c9ea/include/ruckig/path.hpp#L25) is really icing on the cake).

Note: I'm certainly not complaining this is not already supported. Ruckig has been splendid so far, so :+1: and my thanks. Just curious whether that branch has been *left behind* or not :)

pantor, 2021-05-29:
Hi @gavanderhoorn, I haven't worked on it any further and won't get around to it in the next few months. So I definitely keep your interest in mind and will ping you if there are any news.

closed 2021-06-01 by pantor

gavanderhoorn, 2021-06-01:
Ok, thanks.

Not sure whether you need any motivation, but path support would make Ruckig perfect to use with something like [tesseract](https://github.com/ros-industrial-consortium/tesseract) and MoveIt.

> I haven't worked on it any further

could you perhaps describe what is missing, or not working right now?

Is it just bit-rot wrt current `HEAD`?

pantor, 2021-06-01:
Oh, I think you're overestimating that branch :wink: The only part that is implemented is the path generation, but the complete time-parametrization is missing. And as this is the hard part, it really requires a lot of work (and research). Moreover, the algorithms of waypoint- and path-based OTG would differ significantly, so I'm not sure how much I could benefit from waypoint-based Ruckig. Don't have too high hopes - that branch will be research for a long time :laughing: 

Besides this issue, If there is anything I can do to help integrating Ruckig into MoveIt or tesseract, I would be happy to help.