pantor/ruckig #27: No trajectory - (Reached target position in -9.25596e+61 [s].)

Danielxyz123, 2021-05-28:
There is no trajectory with these input parameters  (for example):
   
    Ruckig<1> otg {0.001};
    InputParameter<1> input;
    OutputParameter<1> output;

    // Set input parameters
    input.max_velocity = {2000};
    input.max_acceleration = {18000};
    input.max_jerk = { 190000};

    input.current_position = {100};
    input.current_velocity = {0};
    input.current_acceleration = {0};

    input.target_position = { 1000};
    input.target_velocity = {0.0};
    input.target_acceleration = {0.0};

pantor, 2021-05-28:
Hi @Danielxyz123,

your trajectory is (currently) outside of Ruckigs region of numeric stability. While I haven't published the exact input requirements for numeric stability (yet), I plan to do so soon.

In the meantime, you can normalize your input with a constant factor to get input values closer to 1. So this trajectory with a factor of `0.01` is working fine:
```
input.max_velocity = {20};
input.max_acceleration = {180};
input.max_jerk = { 1900};

input.current_position = {1};
input.current_velocity = {0};
input.current_acceleration = {0};

input.target_position = { 10};
input.target_velocity = {0.0};
input.target_acceleration = {0.0};
```

Afterwards, you can just scale your output with `1/0.01`. This input normalization was even integrated in Ruckig once, so maybe I'll add it back.

pantor, 2021-05-28:
Anyway, I've found the root cause of this error and fixed it in the latest commit on the master branch. Can you confirm?

Also note to check the result of the `update` method. In case of errors, the values of the output class (e.g. the duration in the title of this issue) are invalid.

closed 2021-05-28 by pantor