pantor/ruckig #28: How to use Ruckig in C++11

personal-fork-robot, 2021-06-01:
I want to use Ruckig in C++11,and when i build the package,some error occurs.Here are some questions and solutions.

I've done the following:

1. Using std::optional in a C++11 context:use this header: https://github.com/TartanLlama/optional.It is the equivalent of std::optional.When I finished this, it don 't make any mistakes about "std::optional"

2. lambda auto parameters.
https://github.com/pantor/ruckig/blob/5ad83c019b35960294efd04387b8ed3cd9210763/include/ruckig/block.hpp#L89
I replaced `[](auto& a, auto& b)`with `[](const Profile& a, const Profile& b)`, and it successfully passed.

If it have any unknown questions and hidden errors, please let me know.Thx!

pantor, 2021-06-01:
Hi @personal-fork-robot,

Ruckig makes use of several C++17 features, but I don't think that there would be some hidden errors. If it compiles in C++11, it should be fine! Please note that I really recommend a compiler supporting C++17, so perhaps it's easier to update the compiler before downgrading Ruckig?

Following steps come to my mind:
 - As you've written, replace `std::optional` with some polyfill implementation.
 - Some `auto` parameters would indeed be replaced manually.
 - Ruckig uses `if constexpr`, which would probably the hardest to replace it in C++11. As a start, you could just drop the `constexpr`, resulting in a (very slightly) higher runtime duration. I don't know the exact support of `constexpr` in C++11, so maybe it needs to be dropped for static variables as well.
 - There are some C++17 structured bindings in the codebase.

Feel free to complement this list if I forgot something.

personal-fork-robot, 2021-06-02:
Thank you for your quick feedback @pantor . 
Those are the only two problems I've had for the time being，and `if constexpr` just results in some warnnings in C++11 but not error. Maybe I'll deal with those warnings later, but for now it doesn't affect the program very much.

pantor, 2021-06-02:
Just one remark: `if constexpr` is a [C++17 feature](https://en.cppreference.com/w/cpp/language/if). However, as with all modern C++ features, its support depends on the [exact compiler version](https://en.cppreference.com/w/cpp/compiler_support).

closed 2021-06-02 by pantor

pantor, 2021-06-16:
There is now a [script](https://github.com/pantor/ruckig/blob/master/patch-c%2B%2B11.sh) for patching Ruckig for C++11.

personal-fork-robot, 2021-06-20:
Thx a lot：）