pantor/ruckig #34: at_time feature in python

ljarin, 2021-06-15:
Hi, and thanks for your library!! I am having trouble getting a sample from the trajectory at a specific time, and wanted to ask if I am using the interface correctly. This is an example of how I am calling the library. Thanks in advance.
```
from ruckig import InputParameter, OutputParameter, Result, Ruckig
import numpy as np

num_dims = 2
start_state = [1, 1, 1, 1, 1, 1]
end_state = [2, 2, 2, 2, 2, 2]
max_state = [100, 100, 100, 100]
t = .3

inp = InputParameter(num_dims)
inp.current_position = start_state[:num_dims]
inp.current_velocity = start_state[num_dims:2 * num_dims]
inp.current_acceleration = start_state[2*num_dims:3 * num_dims]

inp.target_position = end_state[:num_dims]
inp.target_velocity = end_state[num_dims:2 * num_dims]
inp.target_acceleration = end_state[2*num_dims:3 * num_dims]

inp.max_velocity = np.repeat(max_state[1], num_dims)
inp.max_acceleration = np.repeat(max_state[2], num_dims)
inp.max_jerk = np.repeat(max_state[3], num_dims)

ruckig = Ruckig(num_dims, 0.05)
out = OutputParameter(num_dims)
res = ruckig.update(inp, out)
print(res)

output = OutputParameter(num_dims)
out.trajectory.at_time(t, output.new_position, output.new_velocity, output.new_acceleration)
print(output.new_position) # These are all 0
print(output.new_velocity)
print(output.new_acceleration)
```

pantor, 2021-06-15:
Hi @ljarin, you're right, there are some problems with the Python wrapper for the `at_time` function. I've now changed the wrapper to return the values instead:
```
new_position, new_velocity, new_acceleration = out.trajectory.at_time(t)
```

ljarin, 2021-06-15:
That worked for me, thanks!

closed 2021-06-15 by ljarin