pantor/ruckig #35: Issue with cpp11 patch

Danielxyz123, 2021-06-17:
Visual Studio tells me that 

"[[maybe_unused]]" 
 for example in the following line

 inline bool check([[maybe_unused]] double tf, double jf, double vMax, double vMin, double aMax, double aMin) {

ist not available for cpp11. Or is it my fault?

pantor, 2021-06-17:
All errors should be fixed now, at least using GCC version 5 on Ubuntu 18.04 (as this is tested now in CI). Can you confirm this?

Danielxyz123, 2021-06-18:
Thx a lot. Yes it works in Visual Studio (Win), but only without compiler WX flag...

There are Error like:

Warnung	C4127	Bedingter Ausdruck ist konstant	

For Example in the following Line (line 56 in profile.hpp)

       if (limits == Limits::ACC0) {

pantor, 2021-06-18:
Ok, I've now removed the flags for treating warnings as errors for the C++ patch. Unfortunately, I don't have the time to fix the warnings itself.

Danielxyz123, 2021-06-18:
👍

closed 2021-06-18 by Danielxyz123

Danielxyz123, 2021-07-21:
Hi Pantor, i only want so give some feedback. It's working on the B&R PLC with the cpp11 patch! It uses the GCC Compiler 6.3.0. The performance seems to be good. Great work 👍