pantor/ruckig #37: Python bindings function PerDOF::append returns pointer to temporary variable data

paolo-mastrapasqua, 2021-06-18:
I'm trying to play with the library on windows using the bindings for python. 
Building was straightforward, but executing `examples/position.py`, I ran into this error message:
"ImporteError: generic_type: cannot initialize type "": an object with that name" 

It ended up being the following function:
```
template<size_t MAX>
struct PerDOF {
    static const char* append(std::string name, size_t DOFs) {
        return (name.append(std::to_string(DOFs))).c_str();
    }
//...
};
```
as you can see the variable `name` is temporary, and the function returns a pointer to its internal storage, that is temporary too, and it get deallocated by the time it gets used.

pantor, 2021-06-18:
Hi @paolo-mastrapasqua, thanks for pointing that out! PR #38 should fix this issue. Can you confirm?

paolo-mastrapasqua, 2021-06-18:
👍, I did the same.

closed 2021-06-18 by pantor