pantor/ruckig #40: pip install fails

mdhom, 2021-06-24:
Hey folks,

i am trying to install the python package in my Windows Terminal. When running `pip3 install ruckig`, the following log occurs:
```
PS C:\s> python3 -m pip install ruckig
Collecting ruckig
  Using cached ruckig-0.3.0.tar.gz (7.5 kB)
Using legacy 'setup.py install' for ruckig, since package 'wheel' is not installed.
Installing collected packages: ruckig
    Running setup.py install for ruckig ... error
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\xxx\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\xxx\\AppData\\Local\\Temp\\pip-install-gz9rcxeg\\ruckig_c44f60f20bde4bb980dfa7cbaeb39be1\\setup.py'"'"'; __file__='"'"'C:\\Users\\xxx\\AppData\\Local\\Temp\\pip-install-gz9rcxeg\\ruckig_c44f60f20bde4bb980dfa7cbaeb39be1\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\xxx\AppData\Local\Temp\pip-record-badakedz\install-record.txt' --single-version-externally-managed --user --prefix= --compile --install-headers 'C:\Users\xxx\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\Include\ruckig'
         cwd: C:\Users\xxx\AppData\Local\Temp\pip-install-gz9rcxeg\ruckig_c44f60f20bde4bb980dfa7cbaeb39be1\
    Complete output (43 lines):
    running install
    running build
    running build_ext
    Traceback (most recent call last):
      File "c:\python39\lib\runpy.py", line 197, in _run_module_as_main
        return _run_code(code, main_globals, None,
      File "c:\python39\lib\runpy.py", line 87, in _run_code
        exec(code, run_globals)
      File "C:\Users\xxx\AppData\Roaming\Python\Python39\Scripts\cmake.exe\__main__.py", line 4, in <module>
    ModuleNotFoundError: No module named 'cmake'
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\xxx\AppData\Local\Temp\pip-install-gz9rcxeg\ruckig_c44f60f20bde4bb980dfa7cbaeb39be1\setup.py", line 65, in <module>
        setup(
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\__init__.py", line 153, in setup
        return distutils.core.setup(**attrs)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\core.py", line 148, in setup
        dist.run_commands()
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\dist.py", line 985, in run_command
        cmd_obj.run()
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\command\install.py", line 61, in run
        return orig.install.run(self)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\command\install.py", line 546, in run
        self.run_command('build')
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\dist.py", line 985, in run_command
        cmd_obj.run()
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\command\build.py", line 135, in run
        self.run_command(cmd_name)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\dist.py", line 985, in run_command
        cmd_obj.run()
      File "C:\Users\xxx\AppData\Local\Temp\pip-install-gz9rcxeg\ruckig_c44f60f20bde4bb980dfa7cbaeb39be1\setup.py", line 24, in run
        out = subprocess.check_output(['cmake', '--version'])
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 424, in check_output
        return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 528, in run
        raise CalledProcessError(retcode, process.args,
    subprocess.CalledProcessError: Command '['cmake', '--version']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\xxx\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\xxx\\AppData\\Local\\Temp\\pip-install-gz9rcxeg\\ruckig_c44f60f20bde4bb980dfa7cbaeb39be1\\setup.py'"'"'; __file__='"'"'C:\\Users\\xxx\\AppData\\Local\\Temp\\pip-install-gz9rcxeg\\ruckig_c44f60f20bde4bb980dfa7cbaeb39be1\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\xxx\AppData\Local\Temp\pip-record-badakedz\install-record.txt' --single-version-externally-managed --user --prefix= --compile --install-headers 'C:\Users\xxx\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\Include\ruckig' Check the logs for full command output.
```
I previously ran `pip3 install cmake` because it was complaining that there was no cmake.

Thanks for your help!

mdhom, 2021-06-24:
I now installed package wheel, i basically get the same log now:
https://pastebin.com/M8ZXyNiF
It seems to not find CMakeLists.txt

pantor, 2021-06-24:
Hi @mdhom, I hope that this issue is fixed with the `0.3.1` release, can you try it again?

I've added the missing files to `Manifest.in` to be included in the Python source package.

mdhom, 2021-06-24:
yes, that particular problem was fixed, thanks. Now it is a different error:
```
 ERROR: Command errored out with exit status 1:
   command: 'C:\Users\MaxDhom\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\MaxDhom\\AppData\\Local\\Temp\\pip-install-rc305p68\\ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63\\setup.py'"'"'; __file__='"'"'C:\\Users\\MaxDhom\\AppData\\Local\\Temp\\pip-install-rc305p68\\ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\MaxDhom\AppData\Local\Temp\pip-wheel-jj7h8e7k'
       cwd: C:\Users\MaxDhom\AppData\Local\Temp\pip-install-rc305p68\ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63\
  Complete output (56 lines):
  running bdist_wheel
  running build
  running build_ext
  -- Building for: NMake Makefiles
  -- The CXX compiler identification is unknown
  CMake Error at CMakeLists.txt:4 (project):
    The CMAKE_CXX_COMPILER:

      cl

    is not a full path and was not found in the PATH.

    To use the NMake generator with Visual C++, cmake must be run from a shell
    that can use the compiler cl from the command line.  This environment is
    unable to invoke the cl compiler.  To fix this problem, run cmake from the
    Visual Studio Command Prompt (vcvarsall.bat).

    Tell CMake where to find the compiler by setting either the environment
    variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
    to the compiler, or to the compiler name if it is in the PATH.

  -- Configuring incomplete, errors occurred!
  See also "C:/Users/MaxDhom/AppData/Local/Temp/pip-install-rc305p68/ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63/build/temp.win-amd64-3.9/Release/CMakeFiles/CMakeOutput.log".
  See also "C:/Users/MaxDhom/AppData/Local/Temp/pip-install-rc305p68/ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63/build/temp.win-amd64-3.9/Release/CMakeFiles/CMakeError.log".
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "C:\Users\MaxDhom\AppData\Local\Temp\pip-install-rc305p68\ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63\setup.py", line 65, in <module>
      setup(
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\__init__.py", line 153, in setup
      return distutils.core.setup(**attrs)
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\core.py", line 148, in setup
      dist.run_commands()
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\Users\MaxDhom\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\wheel\bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\command\build.py", line 135, in run
      self.run_command(cmd_name)
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\Users\MaxDhom\AppData\Local\Temp\pip-install-rc305p68\ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63\setup.py", line 36, in run
      self.build_extension(ext)
    File "C:\Users\MaxDhom\AppData\Local\Temp\pip-install-rc305p68\ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63\setup.py", line 61, in build_extension
      subprocess.check_call(['cmake', ext.sourcedir] + cmake_args, cwd=self.build_temp)
    File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 373, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', 'C:\\Users\\MaxDhom\\AppData\\Local\\Temp\\pip-install-rc305p68\\ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=C:\\Users\\MaxDhom\\AppData\\Local\\Temp\\pip-install-rc305p68\\ruckig_8694ef01a2a642bb8a4bb55bd2cf5b63\\build\\lib.win-amd64-3.9\\', '-DPYTHON_EXECUTABLE=C:\\Users\\MaxDhom\\AppData\\Local\\Microsoft\\WindowsApps\\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\\python.exe', '-DEXAMPLE_VERSION_INFO=0.3.1', '-DCMAKE_BUILD_TYPE=Release', '-DBUILD_PYTHON_MODULE=ON', '-DBUILD_SHARED_LIBS=OFF', '-DCMAKE_POSITION_INDEPENDENT_CODE=ON']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for ruckig
```

pantor, 2021-06-25:
Do you have a C++ compiler, e.g. Visual Studio Compiler, installed on your system?

Currently, the Windows Python package is not pre-compiled (in contrast to the Linux and MacOS packages), so pip will download the source files and tries to compile it on your machine. I would like to upload pre-compiled Windows packages too, so maybe I'll look into that soon.

mdhom, 2021-06-25:
yes i have. i added `C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/SDK/ScopeCppSDK/vc15/VC/bin/cl.exe` to PATH, now i get the following error: 
```
 The C++ compiler

    "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/SDK/ScopeCppSDK/vc15/VC/bin/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/s/github_global/ruckig/build/CMakeFiles/CMakeTmp

    Run Build Command(s):nmake -f Makefile /nologo cmTC_ef2e0\fast &&   "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\SDK\ScopeCppSDK\vc15\VC\bin\nmake.exe"  -f CMakeFiles\cmTC_ef2e0.dir\build.make /nologo -L                  CMakeFiles\cmTC_ef2e0.dir\build
    Building CXX object CMakeFiles/cmTC_ef2e0.dir/testCXXCompiler.cxx.obj
        C:\Users\MaxDhom\AppData\Roaming\Python\Python39\site-packages\cmake\data\bin\cmake.exe -E cmake_cl_compile_depends --dep-file=CMakeFiles\cmTC_ef2e0.dir\testCXXCompiler.cxx.obj.d --working-dir=C:\s\github_global\ruckig\build\CMakeFiles\CMakeTmp --filter-prefix="Note: including file: " -- C:\PROGRA~2\MICROS~3\2019\COMMUN~1\SDK\SCOPEC~1\vc15\VC\bin\cl.exe @C:\Users\MaxDhom\AppData\Local\Temp\nm68F9.tmp
    testCXXCompiler.cxx
    Linking CXX executable cmTC_ef2e0.exe
        C:\Users\MaxDhom\AppData\Roaming\Python\Python39\site-packages\cmake\data\bin\cmake.exe -E vs_link_exe --intdir=CMakeFiles\cmTC_ef2e0.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests -- C:\PROGRA~2\MICROS~3\2019\COMMUN~1\SDK\SCOPEC~1\vc15\VC\bin\link.exe /nologo @CMakeFiles\cmTC_ef2e0.dir\objects1.rsp @C:\Users\MaxDhom\AppData\Local\Temp\nm6948.tmp
    RC Pass 1: command "rc /fo CMakeFiles\cmTC_ef2e0.dir/manifest.res CMakeFiles\cmTC_ef2e0.dir/manifest.rc" failed (exit code 0) with the following output:
    Das System kann die angegebene Datei nicht findenNMAKE : fatal error U1077: 'C:\Users\MaxDhom\AppData\Roaming\Python\Python39\site-packages\cmake\data\bin\cmake.exe' : return code '0xffffffff'
    Stop.
    NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\SDK\ScopeCppSDK\vc15\VC\bin\nmake.exe"' : return code '0x2'
    Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)
```

mdhom, 2021-06-25:
ah and when trying to install from pip, i now get this error:
```
 CMake Error at CMakeLists.txt:73 (find_package):
    By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
    asked CMake to find a package configuration file provided by "pybind11",
    but CMake did not find one.

    Could not find a package configuration file provided by "pybind11"
    (requested version 2.6) with any of the following names:

      pybind11Config.cmake
      pybind11-config.cmake

    Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
    "pybind11_DIR" to a directory containing one of the above files.  If
    "pybind11" provides a separate development package or SDK, be sure it has
    been installed.
```

pantor, 2021-06-25:
The first error log is a little bit over my head (and doesn't seem to be related to Ruckig)...

The second error is just that [pybind11](https://github.com/pybind/pybind11) could not be found, which is used by Ruckig for the Python wrapper. You'll need to install that first.

So I think i should really look into a pre-compiled Python package for Windows :wink:

mdhom, 2021-06-25:
> So I think i should really look into a pre-compiled Python package for Windows 😉.
That sounds great!

i had already installed `pybind11` via pip. i read that error message that it can't find some config file, not the package itself...

pantor, 2021-06-25:
Regarding your edit: The compiled python module is usually called something like `ruckig.cp39-win_amd64.pyd` on Windows and should be somewhere in the build directory. However it seems that your log is just the output of the CMake configuration step, not of the build process itself. But I don't know about Windows, so it might be different there.

mdhom, 2021-06-25:
i got it working now:
1. install pybind:
`pip3 install pybind11`
2. add pybind directory to system variables. name=pybind11_DIR, value=
`C:\Users\xxx\AppData\Roaming\Python\Python38\site-packages\pybind11`
3. `pip3 install ruckig`

Maybe you can hint that in the readme?
Thanks for your help, i can't wait to play with ruckig!

mdhom, 2021-06-25:
ah damn it, i was too fast. installation completed, but python can't find the package now. in my site-packages directory, i can only see the dist-info directory:
[image]

pantor, 2021-06-25:
Ok, pre-compiling the packages was much easier than I thought. 

Can you install Ruckig again via pip by `pip uninstall ruckig` and then `pip install ruckig` again?

mdhom, 2021-06-25:
it installed, but package can't be found still.

mdhom, 2021-06-25:
[image]
it "feels" wrong that there is no `ruckig` directory. An in Relase directory is a file named `ruckig.cp39-win_amd64.pyd`

mdhom, 2021-06-25:
i must admit, i don't have much experience with python packages. i just tried renaming the Release folder to ruckig, and now i don't get a `ModuleNotFoundError` but a `ImportError`:
```
ImportError: cannot import name 'InputParameter' from 'ruckig' (unknown location)
```

pantor, 2021-06-25:
Can you check again by installing version 0.3.3 via pip?

mdhom, 2021-06-25:
works perfectly on windows now! 
```
3.300   -0.044  -0.987  -1.000
3.350   -0.036  -0.993  -1.000
3.400   -0.028  -0.997  -1.000
3.450   -0.018  -0.999  -1.000
3.500   -0.009  -1.000  -1.000
3.550   0.001   -1.000  -1.000
Calculation duration: 94.0 [µs]
Trajectory duration: 3.5441 [s]
```

i am now trying to install it on raspberry pi, and it is failing. had to install `wheel`, and `cmake`, now i get the pybind errors:
```
 Could not find a configuration file for package "pybind11" that is
    compatible with requested version "2.6".

    The following configuration files were considered but not accepted:

      /usr/lib/cmake/pybind11/pybind11Config.cmake, version: 2.2.4
      /lib/cmake/pybind11/pybind11Config.cmake, version: 2.2.4
```

pantor, 2021-06-25:
I would recommend to install pybind11 by cloning the repo:
```
git clone https://github.com/pybind/pybind11.git
cd pybind11
mkdir build
cd build
cmake ..
sudo make install
```

pantor, 2021-06-25:
Great that it works on Windows now!

closed 2021-06-25 by pantor

mdhom, 2021-06-25:
> I would recommend to install pybind11 by cloning the repo:
> 
> ```
> git clone https://github.com/pybind/pybind11.git
> cd pybind11
> mkdir build
> cd build
> cmake ..
> sudo make install
> ```

That helped! Thanks alot!