pantor/ruckig #42: No fixed control cycle

mdhom, 2021-07-05:
Hey,
i want to use your library in a non-realtime environment, so i need to get the status of the trajectory at points with flexible distance in time to each other. I am not sure how to use your (python-)library here. I could imagine that using the `Trajectory.at_time` i could accomplish this, but then i am not sure how to implement changes in `target_position`, `target_velocity` on the fly.
My thoughts would generate something like that:
1. run `update` on Ruckig instance once with the target values of the beginning
2. cyclically run `at_time` as long as target values haven't changed
3. whenever target values change, run `update` once
4. proceed with step 2

Could you give me a little hint how to setup this best?
Thanks!

pantor, 2021-07-05:
Yes, that should be good. Just as you've written, you need to call `update` after changing the input values, and then you can use the trajectory duration and the `at_time` method of the output parameters. Note that in Python the `at_time` method returns the new kinematic state as a tuple instead of using references like the C++ version.

mdhom, 2021-07-05:
ok perfect, thanks. just another question: the duration of the trajectory i get after calling update on the fly, is it the new total duration (including all the time passed since the first call)?

pantor, 2021-07-05:
Yes, it is the total duration from current to target state as in the input parameters.

mdhom, 2021-07-05:
ok great, thanks. if you're ok, i'd like to provide an extra python example with this behaviour the next days?

pantor, 2021-07-05:
Sure, just open a PR for that.

closed 2021-07-05 by pantor