pantor/ruckig #44: Wrong behaviour when changing max_velocity on the fly

mdhom, 2021-07-07:
Hi,

i was testing around with my python on the fly example (see open PR) and discovered the following behaviour:
[image]
What is happening: i start a trajectory with `target_position=10` and `max_velocity=3`. After position is greater than 0.7, i update the trajectory with new InputParameters, where `max_velocity=1`. I would expect, that velocity goes directly down to 1, but instead (as you can see in the trace) the velocity "overshoots" the new target_velocity and goes below 1! I do not understand why it behaves like this, i'd expect it to move along the path painted in red.

I can even create a situation where something totally strange happens, the velocity actually changes its sign for some time, which i think should not happen at all:
[image]

Can you help me with what i'm doing wrong here?

You can find the example source for that output [here](https://pastebin.com/1gbQcmQg).

pantor, 2021-07-07:
Hi @mdhom,

this is intendend behaviour: Ruckig will first and foremost try to get the kinematic state within the allowed limits as fast as possible. Only then a time-optimal trajectory is calculated. 

See #17 for a related discussion. I would like to have an option that disables this behaviour, however it would require a few more equations in step 1 of Ruckig.

mdhom, 2021-07-07:
oh damn, that makes the library quite unusable for me. I am justing moving a car with the trajectory, Imagine how stupid it looks when the car moves backwards although it only needs to change velocity...

pantor, 2021-07-08:
That's unfortunate to hear. I've now opened issue #45.

closed 2021-07-08 by pantor