pantor/ruckig #45: Soft Constraints

pantor, 2021-07-08:
Regarding #17 and #44, we should introduce an option to interpret the kinematic constraints as "soft" ones. Then, we don't want to get within the kinematic constraints as fast as possible, but to avoid overshooting the limits.

This requires to disable the braking trajectory and to add the `UDUD` profiles in step 1. I'm not yet sure how to guarantee a solution in this case.

Danielxyz123, 2023-01-04:
Hi Pantor,

is this feature already implemented in the latest version? (jerk minimizing profiles..?) 

thx for your great work!

pantor, 2023-01-06:
No, the jerk minimizing profiles are independent of the soft constraints here. Thanks!

abernasconi90, 2023-10-17:
Hello Pantor,
I am integrating your library in a project and I am facing the same behaviour of overshooting when changing `max_velocity` on the fly. 
I would ask if there are any update on this topic and if there are also some workaround at the moment?
I tried to set the `min_velocity` parameter equal to zero which can avoid backward movement but I cannot set it to higher value (to decrease more the overshoot) since it may be in conflict with `target_velocity`. Any suggestion?