pantor/ruckig #46: Template makes it difficult to use Ruckig as a class member

AndyZe, 2021-07-08:
I tried to use Ruckig as a class member like this:

`std::unique_ptr<ruckig::Ruckig<6 /* num dof */>> ruckig_;`

A challenge is, I don't know the "num dof" beforehand. So I had to hard-code it to 6 for now. Could you refactor this somehow so that the constructor takes "num_dof" as an argument?

This issue could make it hard to use Rucking in MoveIt where all numbers of DOF are possible.

pantor, 2021-07-08:
Hi @AndyZe, PR #47 should solve your issue. It is already in a good state, however I‘ll add some more tests and do some benchmarking before merging.

pantor, 2021-07-12:
#47 is now merged, so I'll close this issue. If there is anything I can do for integrating Ruckig into MoveIt, I'd be happy to help!

closed 2021-07-12 by pantor

AndyZe, 2021-07-12:
Great!

Right now, MoveIt2 usually parameterizes the trajectory with TOTG. This is acceleration-limited but not jerk-limited.

To my understanding, it would be a good idea to (optionally) pass the trajectory through Ruckig next. Jerk limits would be applied here. Does that make sense to you?

What we would need to do, then, is wrap Ruckig as a plugin just like [TOTG](https://github.com/ros-planning/moveit2/tree/main/moveit_core/trajectory_processing/src). It would be awesome if you can do this! Otherwise we'll get to it soon.