pantor/ruckig #49: Max Accelerations seem to be ignored

pbeeson, 2021-07-16:
I was testing out Ruckig (nice package BTW), and noticed that I get different results if I recreate the Ruckig data structure for each new problem to solve rather than overwriting the input/target positions, velocities, and accelerations.  I'm sorry I can't be more help than that right now  It's late here, and I wanted to file this before stopping for the day.  I'm sure you could reproduce by running some predetermined set of inputs/outputs  in a loop with the Ruckig data structure inside/outside of the loop.

For me, Its as simple as if the for loop below is at the bottom of the snipp or at the top, where I get different results on repeated solves.
```
   Ruckig<6> otg {0.001};
   InputParameter<6> input;
   OutputParameter<6> output;
   
   // Set input parameters
   input.max_velocity = maxVelocities;
   input.max_acceleration = maxAccelerations;
   input.max_jerk = max_joint_jerk_;

   for (std::size_t ii=0; ii < data.size(); ++ii) {
```

pantor, 2021-07-16:
Hi @pbeeson!

When creating the `InputParameter` instance, not all its members are initialized. In particular, `current_position`, `target_position`, and the `max_<...>` limits need be set explicitly. I mean otherwise the whole position-controlled trajectory generation wouldn't make sense, but I'm also open to initialize them all to zero. Those uninitialized values might explain the non-deterministic output. Can you check that this is not the case here? Otherwise, Ruckig should definitely be deterministic.

gavanderhoorn, 2021-07-16:
@pantor: what about initialising all fields to something like a quiet NaN, and then checking they have been properly initialised by the user when they are about to be used?

Using a sentinel value could help distinguish between uninitialised and "some random large value".

pantor, 2021-07-16:
That seems to be a good idea, thanks for your input @gavanderhoorn!

pbeeson, 2021-07-16:
I definitely am setting all (known) required values.  All current, target,
and maxes.

So my thought last night was that there is something else (maybe in the
output Trajectory) that isn’t getting reset properly.

pantor, 2021-07-16:
I just played around with the code and your hints, but couldn't reproduce something along the lines of your issue. It would be great if you could share a complete example!

pantor, 2021-07-16:
Oh, and just to make sure that it's not as simple as that: If you would apply your code snippet to the `example.cpp`, this would change the trajectory between the first and second execution because the input parameter is updated while stepping through the trajectory. This is done in those lines:
```
input.current_position = output.new_position;
input.current_velocity = output.new_velocity;
input.current_acceleration = output.new_acceleration;
```

pbeeson, 2021-07-16:
I tracked this down to an indexing bug in my own code that was generating the data.  Weirdly it got past all CI tests, but the tiny randomness in a bad memory lookup was magnified by the Ruckig process.  Thanks.

closed 2021-07-16 by pbeeson