pantor/ruckig #50: Long times with jerks

pbeeson, 2021-07-16:
In my testing of Ruckig, I've seen correct values for trajectory times when I have jerk set to something really large, like 1e9, but in many cases anything less, even if far from achievable the times that come out of Ruckig seem very high.  I created a little example program that perhaps you could look at and maybe I'm just use Rucking incorrectly (please ignore the otg interval time -- the results are the same with a small interval and a loop on otg.update(), but I was trying for speed with this).

Here is the code:
```
#include <iostream>

#include <ruckig/ruckig.hpp>

using namespace ruckig;

int main() {
    // Create instances: the ruckig otg as well as input and output parameters
    Ruckig<6> otg {10000};
    InputParameter<6> input;
    OutputParameter<6> output;

    // Set input parameters
    input.max_velocity.fill(5);
    input.max_acceleration.fill(10);

    input.current_position.fill(0);
    input.current_velocity.fill(0);
    input.current_acceleration.fill(0);

    input.target_position.fill(0);
    input.target_velocity.fill(0);
    input.target_acceleration.fill(0);

    input.target_position[2]=2.46012;
    input.target_velocity[2]=-1.17513;
    input.target_acceleration[2]=-8.76144;

    for (std::size_t jj=500; jj >= 100; --jj) {
    
      input.current_position[2]=2.46374;
      input.current_velocity[2]=-1.14621;
      input.current_acceleration[2]=-9.79698;
      
      input.max_jerk.fill(jj);
    
      // Generate trajectory
      if (otg.update(input, output) < 0) {
        std::cerr << "ERROR on jerk "<<jj<<"\n";
      } else {     
        std::cout << "For jerk "<<jj<<".  Reached target position in " << output.trajectory.get_duration() << " [s]." << std::endl;
      }
    }
}
```

The results show that with high jerk > 350 rad/sec^3, the results are what I expect (and can validate with other software) for "near infinite" jerk.  But notice the huge jump from the smoothly increasing times as jerk is lowered at jerk==335.  There is a huge jump.  The delta acceleration here is 1.03554 (with is starting and ending in the same direction as velocity, which has a delta of -0.02892 rad/sec and a positional change around -4mm).  So I can't for the life of me figure out why a jerk of 335 would increase the total trajectory time from ~3ms to >0.5 seconds.  Any help you can provide on this is appreciated.

```
For jerk 500.  Reached target position in 0.00311789 [s].
For jerk 499.  Reached target position in 0.00311789 [s].
For jerk 498.  Reached target position in 0.00311789 [s].
For jerk 497.  Reached target position in 0.00311789 [s].
For jerk 496.  Reached target position in 0.00311789 [s].
For jerk 495.  Reached target position in 0.00311789 [s].
For jerk 494.  Reached target position in 0.0031179 [s].
For jerk 493.  Reached target position in 0.0031179 [s].
For jerk 492.  Reached target position in 0.0031179 [s].
For jerk 491.  Reached target position in 0.0031179 [s].
For jerk 490.  Reached target position in 0.0031179 [s].
For jerk 489.  Reached target position in 0.0031179 [s].
For jerk 488.  Reached target position in 0.0031179 [s].
For jerk 487.  Reached target position in 0.0031179 [s].
For jerk 486.  Reached target position in 0.00311791 [s].
For jerk 485.  Reached target position in 0.00311791 [s].
For jerk 484.  Reached target position in 0.00311791 [s].
[... 360 lines cut ...]
For jerk 123.  Reached target position in 0.780873 [s].
For jerk 122.  Reached target position in 0.783492 [s].
For jerk 121.  Reached target position in 0.786154 [s].
For jerk 120.  Reached target position in 0.788861 [s].
For jerk 119.  Reached target position in 0.791613 [s].
For jerk 118.  Reached target position in 0.794412 [s].
For jerk 117.  Reached target position in 0.797259 [s].
For jerk 116.  Reached target position in 0.800154 [s].
For jerk 115.  Reached target position in 0.803101 [s].
For jerk 114.  Reached target position in 0.806098 [s].
For jerk 113.  Reached target position in 0.809149 [s].
For jerk 112.  Reached target position in 0.812254 [s].
For jerk 111.  Reached target position in 0.815415 [s].
For jerk 110.  Reached target position in 0.818634 [s].
For jerk 109.  Reached target position in 0.821912 [s].
For jerk 108.  Reached target position in 0.82525 [s].
For jerk 107.  Reached target position in 0.828651 [s].
For jerk 106.  Reached target position in 0.832116 [s].
For jerk 105.  Reached target position in 0.835646 [s].
For jerk 104.  Reached target position in 0.839245 [s].
For jerk 103.  Reached target position in 0.842914 [s].
For jerk 102.  Reached target position in 0.846654 [s].
For jerk 101.  Reached target position in 0.850469 [s].
For jerk 100.  Reached target position in 0.854359 [s].
```

pantor, 2021-07-17:
First of all, I think Ruckig is working fine here, these should be all effects of the underlying trajectory generation problem. I think the best way to explain this jump of the trajectory duration between jerk 335 and 336 is to have a look at the slower trajectory:

[image]

This is now plotted with a jerk of 100 (and a scaling of acceleration and jerk), but the overall trajectory is the same as with jerk 335. The plot shows that the robot passes its target position after around 0.003s, however has too high velocity and too low acceleration there. That's why the robot accelerates and moves in positive direction again (with velocity > 0), so that it can then reach the target state with exact position, velocity and acceleration. In short, the robot needs to "strike out" to reach its target state. Also note that the robot passes its target position three times: First after 0.003s with negative velocity, then at around 0.6s with positive velocity and then in the end (with its negative target velocity).

When you increase the jerk over the threshold of 335, the robot is now able to avoid this detour and is able to accelerate fast enough to reach the target state without passing over its target position. These jumps happen regularly in online trajectory generation, so it's nothing too special here. And of course, lowering the (or introducing a) jerk limit will always slow down the trajectory, and as it is the third derivative its effect can be quite drastic.

pbeeson, 2021-07-17:
Did you mean to post a different graph image?

pantor, 2021-07-17:
It seems to be the correct figure, but the acceleration and jerk are scaled by a factor of `0.1`. Or what doesn't seem to fit?

pbeeson, 2021-07-17:
I see.  I didn't read the original description correctly.  

I guess the issue is really that because you are calculating things so exactly, any small miss of the velocity or acceleration at the first pass through position will require the quintic spline to do its thing.

That makes total sense.  Thanks for the comprehensive answer.

closed 2021-07-17 by pbeeson