pantor/ruckig #51: Question - inequality constraint on the zeroth derivative + polynomial degree

Iko-Git, 2021-07-17:
1- I'd like to kindly ask if I can impose and inequality constraint on the zeroth derivative with ruckig. Basically, what I am aiming to do is to prevent the position function to exceed certain value!
2- Would it be possible to increase the degree of the polynomial?
If there's an example that would be highly appreciated

pantor, 2021-07-17:
Hi @Iko-Git,

(1) Constraints on the position do not make too much sense in the context of Ruckig: The position extrema of the trajectory are very much specified by the current and target state. So it is (more-or-less) your task to set input parameters that don't constraint your position limits. Ruckig makes it easy to do that by using the `trajectory.get_position_extrema()` method.

By the way, with zero initial and target velocity and acceleration, Ruckig guarantees that the position will only be between the initial and target position. Moreover, position constraints only make sense for the synchronization of multiple DoFs, but it is not implemented there (if that was your question).

(2) Ruckig does not support snap-constrained trajectories. While it is theoretically possible to implement a snap-constrained, time-optimal and real-time capable trajectory generator, it would be very tricky to do so.

Iko-Git, 2021-07-17:
Hello Pantor!
Thanks for your comprehensive reply. 
with respect to the position constraints, I am aiming to perform multi agents trajectory generation (as a piecewise polynomials) what requires from me setting inequality constraints on the segments in the polynomials. However, I could figure how with zero initial and target velocity and acceleration, Ruckig guarantees that the position will only be between the initial and target position, which is fine even for the case of multi agents. However, aiming to make sure that multiagents navigation is safe (with no collision) I am trying to set inequality constraint on the zeroth derivative. 
Thanks!

closed 2021-07-19 by pantor