pantor/ruckig #53: Set synchronization and interface for every DOF

Danielxyz123, 2021-07-22:
Hi Pantor,

it would be nice to deactivate the synchronization for every DOF. It could be usefull for some applications.
Furthermore the same thing with the interface (velocity/position)-mode. 
It would be very nice to synchronize for example one of the dofs with velocity and the others with time.

It's not for a "classic" robot application, but i have a lot of use cases where i could need something like this.

What do you think about that.

Thx and many greetings
 
Daniel

pantor, 2021-07-22:
Hi Daniel,

I'm just curious: What is your use-case for this? In principle, this should be easy to implement, although I'm not sure if the additional complexity is worth it...

Danielxyz123, 2021-07-22:
Hi,

for example there is an system of an linear servo axis and an servo droven drum/roll. 
There should be a part of the movement where the velocity of the linear axis and the drum should be the synchronized.
But, the linear axis needs the velocity at a given position (position mode - time synchronized), for the drum is the position not important (velocity mode - time synchronized).... 
Currently those things are usually solved with a set of 5th oder polynoms, it takes a lot of time to get it working.

Another example..
There is a 3 DOF grasping system where one of the DOFs is an conveyor. Is has to feed the material to the grasping system (velocity mode - unsynchronized only this single axis) and in addition is has to be the third DOF while the system is grasping (position mode - time synchronized of all 3 axes).

I think it would be very usefull for a lot of users.. "real-robots" are only one part in the industrial machine buildung.

pantor, 2021-09-23:
The latest commit adds the members `per_dof_control_interface` and `per_dof_synchronization` to the input class:
```.cpp
// Sets the control interface for each DoF individually, overwrites global control_interface
std::optional<Vector<ControlInterface>> per_dof_control_interface; 

// Sets the synchronization for each DoF individually, overwrites global synchronization
std::optional<Vector<Synchronization>> per_dof_synchronization;
```
Thanks for your input!

closed 2021-09-23 by pantor

Danielxyz123, 2021-10-06:
Works fine! Thx :-)