pantor/ruckig #55: Trigger stop

iadonkey, 2021-07-24:
What is the best way to trigger a stop profile. My use case is as follows

* move 2 axes along a previously computed trajectory
* the “machine” operator pushes a stop button
* I want calculate smooth brake trajectory that stops the axes as fast as possible (max_velocity=X, max_acceleration=Y, target_velocity=0, target_acceleration=0) 
* In this case I do not care about the target_position at all, but I can not simply set target_position = current_position

pantor, 2021-07-24:
The [velocity interface](https://pantor.github.io/ruckig/namespaceruckig.html#a47c13b0e731f894979621b17c63cd75c) is exactly for that use-case. You can activate the velocity interface in C++ by setting `input.interface = Interface::Velocity;`, then the target position will be ignored.

iadonkey, 2021-07-24:
Ok, thanks, works like a charm.
Great work btw, thanks!

closed 2021-07-24 by iadonkey