pantor/ruckig #56: Bug in found_time_synchronization evaluation

stefanbesler, 2021-07-24:
Hi,

I am just integrating some of your commit into the TwinCAT port I am working on. I noticed that the signature of the check Method changed (see https://github.com/pantor/ruckig/commit/b61e99908c715a9e30f83643b177c6c61afc80cd), but is not used correctly in the following code segment 

``` cpp
 switch (p.jerk_signs) {
                        case Profile::JerkSigns::UDDU: {
                            if (!p.check<Profile::JerkSigns::UDDU, Profile::Limits::NONE>(t_profile, new_max_jerk[dof], new_max_velocity[dof], new_max_acceleration[dof], new_min_acceleration[dof])) {
                                found_time_synchronization = false;
                            }
                        } break;
                        case Profile::JerkSigns::UDUD: {
                            if (!p.check<Profile::JerkSigns::UDUD, Profile::Limits::NONE>(t_profile, new_max_jerk[dof], new_max_velocity[dof], new_max_acceleration[dof], new_min_acceleration[dof])) {
                                found_time_synchronization = false;
                            }
                        } break;
                    }
```

I think after `new_max_velocity[dof]` there should be a parameter vor the new minimum velocity instead of new_max_acceleration, or am I missing something here?

pantor, 2021-07-25:
You're right, also phase synchronization doesn't work with the Python interface due to uninitialized Vectors just above your snippet.

pantor, 2021-07-25:
This should be fixed with the latest commits.

closed 2021-07-26 by pantor