pantor/ruckig #60: Numerical issue on PLC (Velocity Mode  ->  "ErrorSynchronizationCalculation")

Danielxyz123, 2021-08-04:
Hi Pantor, 

with the following Parameters i got an "ErrorSynchronizationCalculation" on my PLC Hardware as in issue #54. But this case is in velocity mode "**input.interface = ruckig::Interface::Velocity**". The movement should stop..
`
    Ruckig<4> otg {0.002};
    InputParameter<4> input;
    OutputParameter<4> output;
    // Set input parameters
    input.max_velocity          = { 1752, 1593, 1734, 131};
    input.max_acceleration      = { 17520, 17520, 1738, 1738};
    input.max_jerk              = { 175204, 175204, 26006, 17383};

****
    input.current_position      = {357.23, 406.96, 0.01, -8.38};
    input.current_velocity      = {-1244, -1228, 0.0, -17.01};
    input.current_acceleration  = {11816, 11298, 0.0, 0.0};

    input.target_position       = {300, 349, 0.01, -10}; //doesnt't matter in velocity-mode
    input.target_velocity       = {0.0, 0.0, 0.0, 0.0};
    input.target_acceleration   = {0.0, 0.0, 0.0, 0.0};
    input.interface             = ruckig::Interface::Velocity;
`
I changed the threshold in the marked line to **1e-10** and it works fine.

[image]

pantor, 2021-08-04:
This seems familiar... Again, this does work without error on my computer, but I'll set the threshold to `1e-10` later.

closed 2021-08-04 by pantor

Danielxyz123, 2021-08-04:
Thx a lot!