pantor/ruckig #62: glibc-2.34 compatibility

shr-project, 2021-08-18:
SIGSTKSZ is no longer constant with glibc-2.34 causing:
```
ruckig/0.3.3-1-r0/git/test/doctest.h:4299:47: error: size of array 'altStackMem' is not an integral constant-expression
 4299 |         static char             altStackMem[4 * SIGSTKSZ];
      |                                               ^
```

It was already fixed in Catch2, but the code is quite different there now:

https://github.com/catchorg/Catch2/commit/c0d0a50bdb2ae2f749443c0386c2b25379bdbf76#diff-648e0369c7cddc94be0bfb2b3169679550826a6ac3967f250a43e9fe168608d1

pantor, 2021-08-18:
This should be solved by the latest commit. Thanks for pointing out that issue!

shr-project, 2021-08-18:
Builds fine for me now, thanks for super fast fix.

closed 2021-08-18 by shr-project