pantor/ruckig #65: Odd trajectory output when used for path following

BryanStuurman, 2021-08-19:
I've written a path following algorithm around Ruckig to control a robot through a number of waypoints (position, velocity, acceleration) and Ruckig will occasionally give me a suboptimal trajectory that oscillates about the goal position before continuing on course.

Some background: I load Ruckig with the robots initial position (say 0,0 for a 2 DOF), then the target position is loaded with the first waypoint. I iterate ruckig with input=output and target constant until ruckig returns Finished, then I load the next waypoint (input is equal to the last waypoint) and keep on trucking. I've brought plots (position, velocity, acceleration):
[image]
[image]

The second tracks nicely, however the first axis seems to create a sinusoidal trajectory. One thing I've noticed is that the undesirable response comes when I set the planner to use significant amounts of the robot's velocity limit (0.8 in the above example). The same trajectory with a velocity scaling of 0.61 generates a nice trajectory, as seen in the following plots:
[image]
[image]

What can I do to avoid this response? Am I using Ruckig incorrectly, or perhaps is there a bug that I can work around or fix?

I have uploaded the code used to create these trajectories to my area here on github:
https://github.com/BryanStuurman/ruckig_work/blob/main/trajectory.py

pantor, 2021-08-20:
Hi @BryanStuurman,

this doesn't seem like a bug of Ruckig - it's just that Ruckig calculates *only* a time-optimal trajectory from a current state to a single target state (both defined by position, velocity and acceleration) within the kinematic limits. These "oscillations" happen when the target velocity is too high so that it cannot be reached by accelerating within the position difference. Then, the robot strikes out to reach the *exact* target velocity.

So when you are applying Ruckig to path following, you need to calculate the intermediate velocity and accelerations very carefully - and this is not a trivial task. How did you calculate the velocities here? So MoveIt works on integrating Ruckig to smooth trajectories, which seems like a close use case. They mitigate those backward motions by reducing the velocity in discrete steps until they don't occur anymore. Have a look at their code [here](https://github.com/ros-planning/moveit2/blob/535c39ed2091d27ce928698fb895f680cd604a00/moveit_core/trajectory_processing/src/ruckig_traj_smoothing.cpp).

Otherwise, I'm also working on extending Ruckig to support intermediate waypoints for path following, but unfortunately I can't release anything at the moment.

gavanderhoorn, 2021-08-20:
>  I'm also working on extending Ruckig to support intermediate waypoints for path following,

Just wanted to add a :tada: here

grotius-cnc, 2021-08-23:
Hi,

For info, i also made a robot path planner with ruckig.
- https://github.com/grotius-cnc/hal_trajectory

So far i had no problems. The oscsiloscope output (1ms interval) is perfect.

BryanStuurman, 2021-08-23:
With regards to velocity selection, our motion planner generates a collision free path and then assigns velocities to the robot based on the robot's absolute limits multiplied by a scale factor (to give some margin). I mimic that here with my quick python program which generates a sine path and cos velocities (and -sin accelerations). Ultimately our velocities _have_ to be this high or we wont hit our cycle times and the robotic application becomes uneconomical.

So I played around over the weekend and I found if I drop the 2nd axis Ruckig will generate a beautiful trajectory between waypoints 1 and 2, but backtracks on the way to waypoint 1 this time. Given that it finds a "good" solution for the same conditions, I'm not sure I concur with the idea that the target velocity is too high to be feasible. (reference plot below)
[image]

Hopefully this week I'll have some time to be able to dig into the code a bit more and see why these trajectories come out this way - I'm not adverse to adding an additional (sub-optimal?) profile if it gets away from these responses.

In addition I will review moveit's implementation, and some of the others that are popping up in this thread. I'm pleased to see the community all working on similar problems, I bet we will come up with something excellent.

closed 2021-09-06 by pantor

pantor, 2021-09-09:
Just for your information: We now offer a closed-source version of Ruckig that supports **intermediate waypoints** for path following. In comparison to offline approaches like TOTP or TOPP-RA, Ruckig offers: (1) faster trajectory durations due to joint path and time parametrization, (2) real-time capability, and (3) jerk limits. You can find more information at [ruckig.com](https://ruckig.com), and also some examples and docs in the Readme here. More details and comparisons are coming soon!

weishen-kit, 2021-11-18:
> > Hi,
> > For info, i also made a robot path planner with ruckig.
> > 
> > * https://github.com/grotius-cnc/hal_trajectory
> > 
> > So far i had no problems. The oscsiloscope output (1ms interval) is perfect.
> > simplescreenrecorder-2021-08-02_13.18.45.mp4
> 
> So how did you compute the intermediate velocity/accelerations?

Hi,
I am also curious about your question. I have a large number of intermediate waypoints. From now on i can only calculate intermediate velocity inspired by  MPC with some optimization methods, the whole generated trajectory is not strictly optimal but can effectively reduce  the duration. If you make any progress, we can discuss it.

grotius-cnc, 2021-12-05:
Hi,

My solution so far is to use a single Dofs for the 3d path planning.
For every path, like circle, arc or line i do a few calculations, like path lenght etc.

The single Dofs is repeatetly called for: cartesian xyz and euler xyz positions and they are interpolated outside the ruckig code.

Then when the traject 3d tcp position is known, i make usage of a Dofs vector containing all machine joints for forward
kinematics. For inverse kinematics i also use a Dofs setup with a synchronized motion. This i use for G0 moves. 
When doing a inverse kinematics along a straight line, i will use the single Dofs, with my own interpolation.

https://github.com/grotius-cnc/hal-core/releases => https://github.com/grotius-cnc/hal-core/releases/download/1.18/hal-core.tar.gz

In this package you can find how i use the ruckig. 

I am still busy on the trajectory planner to get the optimal end velocity for each path.

thayjes-srivas, 2024-10-30:
@grotius-cnc When you convert the 3D path to 1DOF, how do you ensure the jerk limit is not violated between the segments?