pantor/ruckig #66: Machine accuracy of used HW - PLC  (question)

Danielxyz123, 2021-08-20:
Hi,

sorry that i bother you with that again. I still have sometimes numeric issues on my **realtime plc hw** (Probably if the working trajectory is already time optimal and i do only small changes?)

In general it's working fine with a scaling of (factor 0.01). 

But in an sensor application where i set cyclic new targetpositions, i often get "ErrorExecutionTimeCalculation"...

I calculated the machine accuracy with the following code. The Result of machEps is 1.0842021724855044E-19. Is it an indicator for you if theres a problem with the accuracy in general on my hw?

`double machEps = 1.0;
	
do
machEps /= 2.0;
while ((double) (1.0 + (machEps / 2.0)) != 1.0);`

pantor, 2021-08-20:
I don't think that there is a general problem with the accuracy of your hardware. So first of all, every ErrorExecutionTimeCalculation is (at least when the input values are valid and within the numeric range stated in the Readme) a bug of Ruckig. So please post the input values (or a few typical ones) leading to that error here.

When you are already on a time optimal trajectory and then make slight changes e.g. to your target, that shouldn't necessrily result in a numeric error. However, Ruckig doesn't and cannot guarantee that the same trajectory will again be calculated, as you are exactly on a switching point between different trajectory profiles. When there is for example a two-step approach for calculating the trajectory, it would make sense the reduce the kinematic limits just a bit (e.g. by `1e-12`) the first time, and then increase the kinematic limits to the original values for the second time optimal profile. This way you could make sure that you get the same overall trajectory profile.

Danielxyz123, 2021-08-20:
Thx for your explanation.

Now i found out the problem...  My usage of the offlinecheck with the second instance seems to be wrong. (If the offline check was bad, i didnt' gave these parameters to the online otg)  

Without that check everything is fine. Probably the fault is, that i didt't call that cyclic like the online instance. But i copied the currentValues cyclic to the offline input structure.

`resultOffline = otgExtrema.calculate(newInput, outputTrajectoryExtrema); ` 

Could you make an simple example for the offline check case?

Danielxyz123, 2021-08-23:
Now i did some tests and theres no way to to get this offline check with the "calculate method" running without any errors. Most of the results are positiv, but not all (about 15% with cyclic input changes).
I have no idea why.

```
resultOffline = otgExtrema.calculate(newInput, outputTrajectoryExtrema);
result        = otg.update(input, output);

//copy for Ruckig loop
input.current_position 			= output.new_position; 	
input.current_velocity 			= output.new_velocity; 	
input.current_acceleration 		= output.new_acceleration; 	
newInput = input; 	//copy for offline instance
```

Danielxyz123, 2021-08-25:
Hi,

with the latest commit regarding to issue #69 and #68 this issue is also solved  👍. Thx a lot, great work!

pantor, 2021-08-25:
Ok great!

closed 2021-08-25 by pantor