pantor/ruckig #69: ErrorExecutionTimeCalculation

Danielxyz123, 2021-08-25:
Hi Pantor,

i got an "ErrorExecutionTimeCalculation" with the following Code.

`
#include <iostream>

#include <ruckig/ruckig.hpp>

using namespace ruckig;

int main() {
    // Create instances: the ruckig otg as well as input and output parameters
    Ruckig<2> otg{ 0.002 };
    InputParameter<2> input;
    OutputParameter<2> output;

    ruckig::Result  resultOnline = ruckig::Result::Working;
    unsigned int counterError = 0;
    bool newPosInitDone = false;

    // Set input parameters
    input.max_velocity      = { 32.34, 19.60 };
    input.max_acceleration  = { 245.0, 98.0 };
    input.max_jerk          = { 2450.0, 2450.0 };

    input.current_position = { 5.0, 0.0 };
    input.current_velocity = { 0, 0 };
    input.current_acceleration = { 0, 0 };

    input.target_position = { 0.1, -3.35 };
    input.target_velocity = { 0.0, 0.0 };
    input.target_acceleration = { 0.0, 0.0 };

    // Generate trajectory
    while ((resultOnline = otg.update(input, output)) == Result::Working) {

        if ((output.new_position[0] < 2.5) && (newPosInitDone == false))
        {
            newPosInitDone = true;
            std::cout << "set new pos!!!!" << std::endl;
            input.target_position[0] = 0.0;
        }

        input.current_position = output.new_position;
        input.current_velocity = output.new_velocity;
        input.current_acceleration = output.new_acceleration;

        auto& p = output.new_position;
        auto& v = output.new_velocity;
        auto& a = output.new_acceleration;

        std::cout << "p: " << output.time << " " << p[0] << ' ' << p[1] << std::endl;

    }

    std::cout << " Error count : " << counterError << std::endl;
    if (resultOnline != Result::Finished)
    {
        std::cout << " Last Error  : " << resultOnline << std::endl;
    }
}
`

pantor, 2021-08-25:
Hi @Danielxyz123,

this does seem to work on my hardware again. Could you replace `Ruckig<2>` with `Ruckig<2, true>` and post the exact thrown error message here?

Danielxyz123, 2021-08-25:
Hi,

i had to modify your code to get the message in my Win10 VS System. Or do you mean another message?

Error message:

[ruckig] error in step 1, dof: 1 input:
inp.current_position = [2.46105171562071, -1.740571215512525]
inp.current_velocity = [-22.05435743108223, -16.24482629348391]
inp.current_acceleration = [0, 9.881190028331117]
inp.target_position = [0, -3.35]
inp.target_velocity = [0, 0]
inp.target_acceleration = [0, 0]
inp.max_velocity = [32.34, 19.6]
inp.max_acceleration = [245, 98]
inp.max_jerk = [2450, 2450]

 Error count : 0
 Last Error  : -110

Without try catch block, it throws no error message...

[image]

Danielxyz123, 2021-08-25:
Now, i used the last commit... It works fine!

pantor, 2021-08-25:
That's great to hear! Then it was probably cause by the same issue as #68. Feel free to open new issues if you get more ExecutionTimeCalculation errors.

closed 2021-08-25 by pantor