pantor/ruckig #73: Can I force phase synchronization?

mqnc, 2021-09-28:
I am using ruckig to control our robot. I want to be sure that the interpolation result is a straight line in joint space and I thought that this is what phase synchronization is for. However, I always obtain a curve in joint space, even if I have 0 velocity and acceleration at start and target.

The source says
```
Phase, ///< Phase synchronize the DoFs when possible, else fallback to "Time" strategy
```

but could you elaborate on "when possible"?

Is there a way to enforce phase synchronization even if it makes the result a bit slower?

The alternative I see is to project all limits onto my target direction in joint space and perform a 1D interpolation. Would you recommend this?

pantor, 2021-09-28:
Hi @mqnc,

first of all, please try again with the latest commit! Ruckig was indeed too strict for deciding when phase synchronization is possible, but that should be fixed now.

Anyway, one necessary condition for phase synchronization is that the input and target velocities and accelerations are collinear (which of course is fulfilled with 0 velocity and acceleration). This becomes a sufficient condition when the trajectory does not need to be time-optimal, otherwise the velocity and acceleration limits need to allow phase synchronization as well. Ruckig currently does only consider time-optimal trajectories for phase synchronization, so there is no way to enforce this by now. But I agree this would be useful feature. Projecting all limits onto the target direction and then using Ruckig for 1 DoF is definitely possible.

Btw, you are [IPR](https://www.ipr.kit.edu) alumnus, aren't you?

mqnc, 2021-09-28:
Thanks for the quick response!
Yes I am! :D I will drop you an email.

chibai, 2025-08-28:
Is there any improvement for this case?