pantor/ruckig #79: Velocity is limited but synchronization of limiting DOF is deactivated

Danielxyz123, 2021-10-15:
Hi Pantor,

the code below shows an example with 3 DOF.
DOF one and two have the same parameters. DOF 3 has a lower vMax, target velocity and the synchronization of this DOF is deactivated.
I expected that the constant velocity phase of DOF 1 and 2 would raise to vMax, but it seems to be limited by DOF three...
If i deactivate the synchronization of all DOF, it works as i expect.
Is it a bug?

```
#include <iostream>

#include <ruckig/ruckig.hpp>

using namespace ruckig;

int main() {
    // Create instances: the ruckig otg as well as input and output parameters
    Ruckig<3> otg{ 0.002 };
    Ruckig<3> otg_offline{ 0.002 };
    InputParameter<3> input;
    OutputParameter<3> output;
    OutputParameter<3> newOutput;

    ruckig::Result  resultOnline = ruckig::Result::Working;

    // Set input parameters
    input.max_velocity          = { 125,   125 ,  100 };
    input.max_acceleration      = { 2000,  2000 , 2000 };
    input.max_jerk              = { 20000, 20000 ,20000 };

    input.current_position      = { 0.0, 0.0 , 0.0 };
    input.current_velocity      = { 0.0, 0.0 ,0.0 };
    input.current_acceleration  = { 0.0, 0.0 ,0.0 };

    input.target_position       = { 35, 35 ,35 };
    input.target_velocity       = { 125, 125, 100 };
    input.target_acceleration   = { 0.0, 0.0 , 0.0 };
    input.per_dof_synchronization = { Synchronization::Time, Synchronization::Time, Synchronization::None }; //velocity doesn't raise to vMax 

    // Generate trajectory
    while (true)
    {
        resultOnline = otg.update(input, output);

        input.current_position = output.new_position;
        input.current_velocity = output.new_velocity;
        input.current_acceleration = output.new_acceleration;

        auto& p = output.new_position;
        auto& v = output.new_velocity;
        auto& a = output.new_acceleration;

        std::cout << "p: " << output.time << " " << p[0] << ' ' << p[1] << ' ' << p[2] << std::endl;
        std::cout << "v: " << output.time << " " << v[0] << ' ' << v[1] << ' ' << v[2] << std::endl;

    }
}
```

pantor, 2021-11-02:
This should be fixed with the latest commit. Thanks for your feedback!

Danielxyz123, 2021-11-04:
Hi Pantor,
thx for that fix.

The velocity while synchronization ist working fine now. But there is a new problem since this commit.

The velocity mode per dof doesn't work if the synchronization is "Synchronization::None" for that dof. This is tested on the plc...
Later i'll test it on a pc, and i'll send you the testcode.

Note that the problem isn't existing in the commits before this.

pantor, 2021-11-04:
Oh yeah, I see the problem. Sometimes it's quite tricky...

pantor, 2021-11-08:
I hope that this is fixed now on the master branch. Can you check that?

Danielxyz123, 2021-11-08:
Thx a lot, now everything seems to be fine. 

But the patch script isn't working in that script folder.... I started it from root.

pantor, 2021-11-08:
Perfect.

The patch script is meant to be run from the root directory as done in the [CI](https://github.com/pantor/ruckig/blob/095e9e6b7a4986856e2886e4a073d34ff23101f1/.github/workflows/ci.yml#L36) and written in the Readme.

closed 2021-11-08 by pantor