pantor/ruckig #81: Feature Request: Varying dynamics limits

mrunaljsarvaiya, 2021-11-01:
Is there a plan to integrate varying limits into Ruckig? I.e If I want to move through points A,B,C,D I might want the robot to have different velocity limits between B and C.

pantor, 2021-11-02:
Hi @mrunaljsarvaiya!

moving through intermediate points is the primary feature of the Ruckig Pro Version (in comparison to the Community Version published here). While the current version does not support varying limits, this is already implemented in the upcoming version which will get released in the following weeks. For more information about Ruckig Pro, feel free to visit [ruckig.com](https://ruckig.com) or contact us at [info@ruckig.com](mailto:info@ruckig.com).

pantor, 2021-11-09:
This is now implemented in Ruckig Pro.

closed 2021-11-09 by pantor