pantor/ruckig #84: AttributeError: 'ruckig.OutputParameter' object has no attribute 'pass_to_input'

nkashiri, 2021-11-05:
When running the python examples including the pass_to_input function, it gives the error:
AttributeError: 'ruckig.OutputParameter' object has no attribute 'pass_to_input'

The installation was done using:  pip install ruckig
If it helps: when I run >> dir(ruckig.OutputParameter) 
I recieve: 
['__class__', '__copy__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'calculation_duration', 'degrees_of_freedom', 'new_acceleration', 'new_calculation', 'new_position', 'new_velocity', 'time', 'trajectory', 'was_calculation_interrupted']

pantor, 2021-11-05:
Hi @nkashiri,

unfortunately the docs in the master branch are for the upcoming version. Pip currently installs Ruckig version 0.4.0, so please refer to this [Readme](https://github.com/pantor/ruckig/tree/62ba02edd92ed8ec99dc32a4fa6b66deee70c493) in the meantime. We'll release 0.5.0 very soon!

nkashiri, 2021-11-05:
Thanks @pantor for the quick reply, 
The [Readme](https://github.com/pantor/ruckig/tree/v0.4.0) unfortunately concern only the cpp implementation, so I used the [collection of examples](https://docs.ruckig.com/pages.html) that seems to be concerning the 0.4 version. 
looking forward to the new release

closed 2021-11-14 by pantor