pantor/ruckig #85: question about motion planner

grotius-cnc, 2021-11-11:
Hi,

First i want to thank the writers and contributers of Ruckig.
I used this library intensively for my projects, and i am happy with it.
Here are some examples: https://github.com/grotius-cnc/hal-core
And a video of a ruckig realtime implementation : https://user-images.githubusercontent.com/44880102/127900012-b7c19051-7d22-45dc-b7ee-0a229e1caa60.mp4

I have a gcode program that is interpolated by ruckig, an sich this works ok. See picture for example, but there is a 
implementation problem.

In some cases the machine has to slow down at sharp corners, at P4 and P5 the machine has to slow down to avoid a shooting over.
At P2 there seems to be no problem. The conclusion, when paths are in line, the machine can stay at 100% speed.
If the machine has to make corners, the machine has to slow down.

Can ruckig solve this problem?

My idea was to sample each motion line (in a sort of look ahead function) until there is a minimal (or acceptable) overshoot (Ferror).
 
I was wondering if anyone else had to deal with this problem. And maybe has a better idea?

Thanks.

[image]

pantor, 2021-11-18:
Hi @grotius-cnc,

the Ruckig Community Version is not really suited for path following. For a general case, finding velocities and accelerations given a set of intermediate positions is a difficult and complex task. There is/was also a discussion at #65 that might be of interest here.

If you are talking about *sharp* corners, a jerk-limited trajectory will have to slow down to zero velocity anyway. If some path deviation is ok, your approach would work fine, so this is definitely a way to go in your specific case.

closed 2021-11-20 by pantor

grotius-cnc, 2021-12-05:
Thank you for your comment.
I will work on this issue coming day's. I have been delayed to solve this.

_If some path deviation is ok,_
The strategy of servo controllers is to follow the given coordinates as good as possible. A difference of encoder feedback
and position command is always the case during a run, so far as i know.

Currently i have 2 idea's.

1.  Creating a chunk of c++ code and try to implement senario's with some kind of look ahead function.
This has downsides at runtime. It eats memory and processor pieces, and i am already at ~100%  load.
I also think this calculation has to be done before runtime. So all the optimal "velocity end's" are known.
2.  Creating a Ai fann module and train the machine to optimal results. (difficult course)
At runtime, it has the training set with a few values. I think processor is happy.
something like this, but then complexer  : https://github.com/grotius-cnc/hal_artificial_neural_network_xor2

https://user-images.githubusercontent.com/44880102/144765887-f340a338-94d0-4e9b-b308-38a88b8f102a.mp4

pantor, 2021-12-06:
Looking forward to results from the FANN module!