pantor/ruckig #87: Support ruckig as a catkin package

ags-dy, 2021-11-17:
I see you have a ROS package.xml, but if I I add ruckig to my package.xml and find it as a catkin package, ruckig fails to get built as a dependency.  

I also build ruckig manually with ROS colcon which works, but it doesn't seem to install it in the files in a place that other ROS packages can find.  

Is it possible to support it as a ROS package?

Thanks

gavanderhoorn, 2021-11-17:
I originally contributed the manifest in #10.

> if I I add ruckig to my package.xml and find it as a catkin package, ruckig fails to get built as a dependency.

do you mean you try to add it to `find_package(catkin REQUIRED COMPONENTS ..)`?

If so: there is a difference between a plain-CMake package which can be used as a *system dependency* and has a `package.xml` to help tools like `rosdep` discover it, and a Catkin package.

Only Catkin packages can be added as a `COMPONENT` in a `find_package(catkin ..)` call.

But it's not needed for Ruckig, as it doesn't use any ROS infrastructure.

You should be able to just:

```cmake
find_package(ruckig REQUIRED)
...
target_link_libraries(<your_target> PUBLIC ... PRIVATE ruckig::ruckig)
```

and it should work.

> I also build ruckig manually with ROS colcon which works, but it doesn't seem to install it in the files in a place that other ROS packages can find.

This suggests you are actually using ROS 2, which changes things somewhat.

IIRC, for ROS 2 tools to be able to find plain-CMake packages, you'd have to install a `.dsv` marker file in the appropriate location (so it gets registered in the Ament Index). I'm not exactly sure which parts are required, but see [these lines](https://github.com/ros-industrial/ros_industrial_cmake_boilerplate/blob/e1b189615026f00ef0b46fbb5cf3367a2b1019a9/ros_industrial_cmake_boilerplate/cmake/cmake_tools.cmake#L194-L203).

But I would probably recommend to use Ruckig as a system dependency instead.

ags-dy, 2021-11-18:
Thank you for your quick reply!  I was thrown by the existence of the package.xml and was in catkin mode, and totally missed the non catkin way of doing things.

I am on ROS1 with colcon, and your instructions work

thanks!

closed 2021-11-18 by ags-dy