pantor/ruckig #90: Error in step 1 of trajectory calculation

sarvkindred, 2021-11-24:
Hi @pantor, while trying to generate a trajectory between two points, I'm getting an error at[ step 1](https://github.com/pantor/ruckig/blob/5dcdaa654a5d19ef9d7fd745e38294f5a19605d1/include/ruckig/calculator.hpp#L244)

Runtime Error: [ruckig] error in step 1, dof: 5 input: 
inp.current_position = [0.1586916078246517, 0.8198410238869502, -0.2681910390107047, -0.09903812486505789, -1.795801648113672, 1.250459998705841]
inp.current_velocity = [1.224365429569406, -0.8752573930094795, 0.6074060830631294, -0.7773969322150299, -0.6074782842272427, 2.794826109810867e-08]
inp.current_acceleration = [12.36606404941746, 0, 0, 0, 0, 0]
inp.target_position = [0.35257, 0.7036, -0.20583, -0.18096, -1.85703, 1.25046]
inp.target_velocity = [0, 0, 0, 0, 0, 0]
inp.target_acceleration = [0, 0, 0, 0, 0, 0]
inp.max_velocity = [2.663652051520562, 1.555190266438036, 0.9215599306967612, 1.17514790481581, 0.9107239220283427, 2.794826109810866e-08]
inp.max_acceleration = [49.44190521665187, 41.42429896530291, 54.7869760508845, 73.49472397069871, 72.82659011641964, 133.6267625041427]
inp.max_jerk = [772.5298358235708, 647.254671332858, 856.0465007950703, 1148.355062042167, 1137.915470569057, 2087.918294622123]

Do you have suggestions on what might have gone wrong?

pantor, 2021-11-27:
Hi @sarvkindred!

Unfortunately I'm not able to reproduce this error. I've set up a [Colab notebook](https://colab.research.google.com/github/pantor/ruckig/blob/master/test/demo.ipynb) to make sure we're using the same environment, and Ruckig outputs a trajectory just fine there.

This might be caused by a numeric error, as even very slight changes in the InputParameter might influence the result. Is there any chance that you can find an (slightly adapted) InputParameter within the Colab notebook that outputs your error? Otherwise, make sure to use the latest Ruckig version - there are some fixes for numerical stability in v0.5.0.

Also, there are some numeric constraints on the input. Please find more information in [this section](https://github.com/pantor/ruckig#tests-and-numerical-stability) of the Readme. This is in particular relevant for the very low *max velocity* in the relevant DoF 5 of your example input. Could you try to increase this velocity limit? How do you came up with this velocity limit?

atakan-kanman-kindred, 2022-01-05:
this issue is no longer reproducible on our end either, will create a new issue if anything similar pops up. thanks!