pantor/ruckig #92: Rising output values

Danielxyz123, 2021-12-02:
Hi,

since the showed commit below, the output values are not konstant if ruckig has finished the trajectory. (Test on PLC)

"use integrated instead target values after traj" 

They are rising permanent very slow..
Of course, i also call the update method permently if the trajectory has finished.

pantor, 2021-12-02:
Yeah, that commit changed the behavior to use the integrated final values instead of the given target values, so there might be differences due to numeric reasons. This might not be ideal if you're keep using Ruckig a long time after the trajectory has finished.

On the other hand, this difference is really small (e.g. <10e-12 for the velocity), so from my understanding it should hardly be relevant in real-world use cases. The Readme also states upper bounds for this drift. The alternative is using the given target values, then however jumps in the kinematic state might occur.

Is this a problem for your use case, or just something you've observed?

Danielxyz123, 2021-12-03:
Yes thats the reason... There is the possibility that ruckig is in "idle" for a long time.
And yes, it's not probably not relevant. It looks a little bit strange if all output values are "moving", even if it's a size of 10e-12. Isn't there the possibility to "freeze" them when the trajectory has finished?

pantor, 2021-12-09:
I think the way forward here would be to set the position to the integrated value, and the velocity and acceleration to the target values. This way, moving values would be avoided while still reducing discontinuities in the position at least.

Danielxyz123, 2022-06-22:
It works, thx for help.

closed 2023-04-08 by pantor