pantor/ruckig #93: I can't compile a library for python on raspberry pi

Ruslan2288, 2021-12-02:
[image]
[image]

On raspberry pi, the library is not installed using pip when compiling, it writes the same error as in the photo.

Ruslan2288, 2021-12-02:
Please tell me what I need to do to install it. I installed pybind11, but the compiler doesn't seem to see it

pantor, 2021-12-02:
Alas there are no pre-build binaries for ARM, so it wants to compile the library including the Python wrapper. For that, it needs to find [pybind11](https://github.com/pybind/pybind11). You can either install that globally, or just clone it into the Ruckig directory. 
```
cd ruckig
git clone https://github.com/pybind/pybind11.git
```
CMake should pick Pybind11 up from there.

closed 2021-12-05 by pantor