pantor/ruckig #94: FastAPI access to Pro version: rate limits?

gavanderhoorn, 2021-12-07:
Hi @pantor. Thanks for making the FastAPI access to the Pro version available. That's a really nice early Christmas present :+1: 

Before suggesting people to use this (such as @AndyZe with MoveIt 1 or 2), could you say something as to whether you've implemented any rate limitation? Even off-line use by (for instance) the MoveIt community could result in quite some load on the API server.

pantor, 2021-12-07:
Hey @gavanderhoorn, that's nice to hear :wink:

Currently there are no rate limits in the API. Of course, I only have so many (server) resources that I can provide for free, but if this is something of interest for MoveIt, I would be happy to somehow commit to it. Just to give a rough estimate, a normal server for a few dollars a month should be able to handle 100s to 1000 of requests per second.

I think the first and important part is to make sure that this trajectory generation (and its integration) is useful within MoveIt. Ruckig really works best with as few waypoints as possible, so this is quite a different approach than TOPP-RA or TOTP. Needless to say, there are still a lot use cases where Ruckig results in faster and smoother (and of course jerk-constrained) trajectories. So feel free to test this extensively and we'll see where we'll get then :wink: Looking forward to your feedback!

closed 2021-12-10 by pantor

gavanderhoorn, 2021-12-10:
Thanks for clarifying.

Let's see how things work out then.