pantor/ruckig #95: no smooth transition between trajectory segments despite using DurationDiscretization.Discrete

mqnc, 2021-12-14:
While I was working on this bug report, I discovered that ruckig has an offline trajectory mode which solves all my problems in life. Still, I think this is a bug and tracking it down will make the world a better place.

I want to piece together several trajectories but there were always jumps in the transition points. I attached an example. Sorry the code is so redundant but this was the easiest way to generate the problem from my cpp source.

I noticed that it's always just one parameter that has a problem. I suspect that it's the critical one that determines the duration and you just forgot to slow this one down to the next biggest multiple of dt.

```py
from copy import copy
import matplotlib.pyplot as plt
import numpy as np
from ruckig import Ruckig, InputParameter, OutputParameter, Result
import ruckig

def sample(
		startPosition,
		startVelocity,
		startAcceleration,
		targetPosition,
		targetVelocity,
		targetAcceleration,
		maxVelocity,
		maxAcceleration,
		maxJerk
):
	otg = Ruckig(6, 0.016)
	inp = InputParameter(6)
	out = OutputParameter(6)

	inp.current_position = startPosition
	inp.current_velocity = startVelocity
	inp.current_acceleration = startAcceleration
	inp.target_position = targetPosition
	inp.target_velocity = targetVelocity
	inp.target_acceleration = targetAcceleration
	inp.max_velocity = maxVelocity
	inp.max_acceleration = maxAcceleration
	inp.max_jerk = maxJerk

	inp.synchronization = ruckig.Synchronization.Time
	inp.duration_discretization = ruckig.DurationDiscretization.Discrete

	first_output = None
	traj = [startPosition]
	res = Result.Working
	while res == Result.Working:
		res = otg.update(inp, out)
		out.pass_to_input(inp)
		traj.append(out.new_position)
		if not first_output:
			first_output = copy(out)

	return traj

curve = [[-0.1, -1.2, 2, -0.5, -4.71238898038469, -6.2]]

curve.extend(sample(
	startPosition=[-0.1, -1.2, 2, -0.5, -4.71238898038469, -6.2],
	startVelocity=[0, 0, 0, 0, 0, 0],
	startAcceleration=[0, 0, 0, 0, 0, 0],
	targetPosition=[1.9679526741215, -1.79337540810325, 1.60165402119475, -
					0.742383751689976, -4.42451465459919, -4.63004355951186],
	targetVelocity=[1.4216077453218, -0.373745261217006, -0.436991883620502, -
					0.162165540016975, 0.194020004947316, 1.48964658466271],
	targetAcceleration=[0, 0, 0, 0, 0, 0],
	maxVelocity=[1.9998, 1.9998, 1.9998, 1.9998, 1.9998, 1.9998],
	maxAcceleration=[9.999, 9.999, 9.999, 9.999, 9.999, 9.999],
	maxJerk=[9.999, 9.999, 9.999, 9.999, 9.999, 9.999]
)[1:])

curve.extend(sample(
[... 104 lines cut ...]
					0.0271773718891913, 0.274610306393505, 0.0745633868515529],
	targetAcceleration=[0, 0, 0, 0, 0, 0],
	maxVelocity=[1.1994, 1.1994, 1.1994, 1.1994, 1.1994, 1.1994],
	maxAcceleration=[5.997, 5.997, 5.997, 5.997, 5.997, 5.997],
	maxJerk=[5.997, 5.997, 5.997, 5.997, 5.997, 5.997]
)[1:])

curve.extend(sample(
	startPosition=[0.640183128398815, -0.750769855691266, -
				   1.51940658643282, -2.07063582344055, 3.85679424946687, 2.46344067814976],
	startVelocity=[0.171744687193063, -0.0584947649324608, -0.0205105546487691, -
				   0.0271773718891913, 0.274610306393505, 0.0745633868515529],
	startAcceleration=[0, 0, 0, 0, 0, 0],
	targetPosition=[3.04159265358979, -1.94159265358979, -
					2, -2.64159265358979, 4.71238898038469, 6.2],
	targetVelocity=[0, 0, 0, 0, 0, 0],
	targetAcceleration=[0, 0, 0, 0, 0, 0],
	maxVelocity=[1.1994, 1.1994, 1.1994, 1.1994, 1.1994, 1.1994],
	maxAcceleration=[5.997, 5.997, 5.997, 5.997, 5.997, 5.997],
	maxJerk=[5.997, 5.997, 5.997, 5.997, 5.997, 5.997]
)[1:])

plt.plot(np.diff(curve, axis=0))
plt.show()
```

mqnc, 2021-12-14:
[image]

pantor, 2021-12-17:
Thanks for spotting this! It is now fixed with the latest commit in the master branch.

There was indeed a bug with the time synchronization of the limiting DoF in case of discrete durations.

closed 2021-12-17 by pantor