pantor/ruckig #96: ErrorSynchronizationCalculation goes away by lowering max jerk

fricher, 2021-12-16:
Hi @pantor 

I get an ErrorSynchronizationCalculation with the following input
```
inp.current_position = [0.2473592757796861, 0.2921606775204735, 0.7758663276711127, 0, 3.141592653589793, 0]
inp.current_velocity = [-0.2426115138900957, 0.2200706500820608, -0.01891492763905089, 0, 0, 0]
inp.current_acceleration = [0.01877538437863763, -0.6573642866096158, 0.5, 0, 0, 0]
inp.target_position = [0.2308075286416321, 0.3066442218484541, 0.7733155040940536, 0, 3.141592653589793, 0]
inp.target_velocity = [0, 0, 0, 0, 0, 0]
inp.target_acceleration = [0, 0, 0, 0, 0, 0]
inp.max_velocity = [0.25, 0.25, 0.25, 0.35, 0.35, 0.35]
inp.max_acceleration = [2.5, 2.5, 2.5, 3.5, 3.5, 3.5]
inp.max_jerk = [62.5, 62.5, 62.5, 87.5, 87.5, 87.5]
```
But it works fine with a lower max jerk, e.g.: 
```
inp.max_jerk = [32.5, 32.5, 32.5, 47.5, 47.5, 47.5]
```
Is this expected ?

Thanks for your help !

pantor, 2021-12-17:
Hi @fricher,

no, this definitely smells like a bug. As I'm not able to reproduce this error, I might need some more information:
- Can you make sure that you are using the latest Ruckig version v0.6.0?
- What is the exact error message?

fricher, 2021-12-28:
Hello @pantor , sorry for not answering sooner;

The problem seems to actually come from 
` inp.duration_discretization = ruckig::DurationDiscretization::Discrete; ` I'm sorry I failed to mention this in my first post. Updating to master didn't help.

With delta_time=0.004 and DurationDiscretization::Continuous, update succeeds and the trajectory duration is 0.136745

With DurationDiscretization::Discrete and after enabling throw_error, I get this exception:
```  
what():  [ruckig] error in step 2 in dof: 1 for t sync: 0.140000 input: 
inp.current_position = [0.2473592757796861, 0.2921606775204735, 0.7758663276711127, 0, 3.141592653589793, 0]
inp.current_velocity = [-0.2426115138900957, 0.2200706500820608, -0.01891492763905089, 0, 0, 0]
inp.current_acceleration = [0.01877538437863763, -0.6573642866096158, 0.5, 0, 0, 0]
inp.target_position = [0.2308075286416321, 0.3066442218484541, 0.7733155040940536, 0, 3.141592653589793, 0]
inp.target_velocity = [0, 0, 0, 0, 0, 0]
inp.target_acceleration = [0, 0, 0, 0, 0, 0]
inp.max_velocity = [0.25, 0.25, 0.25, 0.35, 0.35, 0.35]
inp.max_acceleration = [2.5, 2.5, 2.5, 3.5, 3.5, 3.5]
inp.max_jerk = [62.5, 62.5, 62.5, 87.5, 87.5, 87.5]
```  
in calculator_target.hpp:374

pantor, 2021-12-30:
Thanks for clearing that up, I can reproduce this issue now. Can you please try again using the latest commit from the master branch?

closed 2022-01-02 by pantor