pantor/ruckig #102: Is there any way to set a minimum jerk?

bleach31, 2022-02-03:
Thanks for the great tool.

Is there any way to set a different Jerk for acceleration and deceleration?

For example, when generating a trajectory leading to a speed of 25, the same Jerk is used for acceleration and deceleration.

I would like to slow down (or steepen) the gradient of the deceleration independently of the acceleration.

```python
inp.current_position = [0.0]
inp.current_velocity = [0.0]
inp.current_acceleration = [0.0]
inp.target_position = [100.0] # long enough
inp.target_velocity = [25]
inp.target_acceleration = [0.0]
inp.max_velocity = [0.5]
inp.max_acceleration = [4.0]
inp.max_jerk = [3.0]
````

[image]

pantor, 2022-02-06:
Hi @bleach31,

this is currently not possible. While the underlying algorithm of Ruckig would be straight-forward to extend to a minimum jerk, we rejected that idea due to the algorithmic and computational complexity (and vanishing use cases). Out of curiosity, why do you need a minimum jerk for your application?

bleach31, 2022-02-09:
Thank you for your reply.
I understand and the current ruckig is no problem! this is enough great. Thank you for developing this.

So,  I will separate sections depending on jerk@max and jerk@min in advance, then I will call rucking for each section.

> Out of curiosity, why do you need a minimum jerk for your application?

I'm planning to use ruckig to define simple vehicle trajectories. In vehicle, the jerk is important because it affects the human's ride comfortability.

closed 2022-02-09 by bleach31

pantor, 2022-02-09:
Thanks! If you have any commercial interest in that feature, feel free to shoot me a mail at info@ruckig.com.