pantor/ruckig #103: Usage in a 3D world

ethindp, 2022-02-20:
Just looking at things like the Brake profile calculation, and I'm wondering if these are what I think they mean.

* t: time,
* j: Jerk
* aa: acceleration
* v: Velocity
* p: Position

It also seems to me as though these are 2-dimensional. Am I misinterpreting these values?

pantor, 2022-02-21:
Hi @ethindp, your understanding of the variable names is correct. However the 2 dimensions of the array do not correspond the number of DoFs of the system (or the 3D world), but to the number of steps in the (brake) profile. In general, a brake profile might need to apply different jerk values one after another to get into the specified kinematic limits - this is why there are 2 dimensions here. Ruckig itself works with an arbitrary number of DoFs (and in particular with 3, 6 or 7 commonly found in real-world usage).

closed 2022-02-21 by pantor

ethindp, 2022-02-21:
@pantor Thank you! Are all the variables used in the code and their meanings documented in the paper? E.g.: the limits and jerk signs in the profile system, for example? I'm new to trajectory calculation so the more documentation the better. :) It would also help me explain it to others who use it -- I'm integrating it into a game that a group of friends and myself are developing.