pantor/ruckig #104: Flaky oscillations when Ruckig timestep is not constant

AndyZe, 2022-02-21:
With a constant timestep, we get nice behavior out of Ruckig. But if the input has a different timestep at every iteration, we occasionally see oscillations. Got any insight on this?

Constant timestep on the left, dynamic timestep on the right:
[image]

Zoomed on the unexpected oscillation:
[image]

Here's the code used for the dynamic timesteps. It just constructs a new Ruckig instance with a different timestep at every waypoint:

```
      timestep = trajectory.getWayPointDurationFromPrevious(waypoint_idx + 1);
      ruckig_ptr = std::make_unique<ruckig::Ruckig<RUCKIG_DYNAMIC_DOF>>(num_dof, timestep);
```

AndyZe, 2022-02-21:
For the nice constant timestep graph, the timestep was 0.1s. For the wacky dynamic timestep graph, it ranged from 0.03-0.25s.

pantor, 2022-02-22:
Hey @AndyZe! Can you provide me a bit more information here, maybe even a simple code example? Are the figures showing the position over time? Ruckig is not really meant to be used with dynamic time steps, so I'm not exactly sure what's going on here.

pantor, 2022-03-15:
Feel free to reopen, but I think I need more information to help here.

closed 2022-03-15 by pantor