pantor/ruckig #107: A couple questions on usage

ethindp, 2022-02-25:
I'm working on integrating this into a project and I have a couple questions:

1. Should I update the input parameters dynamically? Or are they effectively "goals" -- e.g.: if my current velocity is [2, 2, 3], and the new velocity is [2, 5, 6], would that be a goal to get to, or should I immediately set my velocity to the new velocity? As it currently stands, I immediately pass output to input and don't do anything in the control loop.
2. Is there a procedure/formula for me to calculate the maximum velocity/acceleration/jerk, or is that something I must know beforehand?

pantor, 2022-03-01:
Within the control loop, you just need to use the `output.pass_to_input(input)` method and make use of the `output` parameter (e.g. `new_position`). Otherwise, the target parameters are your goals that you only need to set once (or when you actually want to change them on the fly). So you would set your target velocity to `[2, 5, 6]` and update your control loop until the trajectory has finished (by checking the result parameter).

No, you need to define your kinematic limits for your use case.

pantor, 2022-03-04:
Feel free to reopen if you have any more questions!

closed 2022-03-04 by pantor