pantor/ruckig #108: ErrorSynchronizationCalculation

fricher, 2022-02-25:
Hello @pantor ,

I'm hitting a ErrorSynchronizationCalculation error with these input parameters:
```cpp
ruckig::Ruckig<6> otg{0.008};
ruckig::InputParameter<6> inp;
ruckig::OutputParameter<6> outp;

inp.synchronization = ruckig::Synchronization::Phase;
inp.control_interface = ruckig::ControlInterface::Velocity;

inp.current_position = {0.04899089988984485, -0.3736623957062429, -0.007154243077742151, 0.3549484836962968, -0.7307530792012156, -0.3947326782044243};
inp.current_velocity = {0.1119477497536703, -0.005706738140158095, 0.001348657044096806, 0.1054435897294738, -0.03687579980795278, -0.1371515680506455};
inp.current_acceleration = {-2.943871184141059, 0.1638588832925878, -0.03832918779616604, -3.516112331376623, 1.09743295912854, 4.327530374547349};
inp.target_position = {0, 0, 0, 0, 0, 0};
inp.target_velocity = {0, 0, 0, 0, 0, 0};
inp.target_acceleration = {0, 0, 0, 0, 0, 0};
inp.max_velocity = {6.457718232379019, 5.410520681182422, 7.155849933176751, 9.599310885968812, 9.424777960769379, 17.4532925199433};
inp.max_acceleration = {26.89552377323262, 22.5322006432468, 29.81022362406315, 39.98549316319009, 39.61897402027128, 72.71041663808377};
inp.max_jerk = {224.2224490037115, 187.8497873921497, 248.4650723139127, 333.3055272533571, 330.268654354887, 606.0132228774711};
```

I've actually hit this error several times but when I tried to replicate it with the printed values (inp.to_string()), it would actually succeed... In case it's useful, these parameters were printed after an error, but they work fine on their own:
```cpp
inp.current_position = {0.2383856650790576, -0.3299342630619511, -0.2992587405955631, 0.7421485723672347, -0.5772957742147332, -0.9083338322091556};
inp.current_velocity = {0.1655224369207376, 0.009499080610294053, -0.003004920394338419, 0.1470437982102569, -0.1105344456172633, -0.2372752552458836};
inp.current_acceleration = {-8.615549451013575, -0.2516017650111069, 0.07852697334192837, -4.6830877805969, 3.307791575962221, 7.328339967828701};
```

Thanks a lot for your help !

fricher, 2022-02-25:
With exceptions enabled: 

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  [ruckig] error in step 2 in dof: 1 for t sync: 0.035263 input: 
inp.current_position = [0.04899089988984485, -0.3736623957062429, -0.007154243077742151, 0.3549484836962968, -0.7307530792012156, -0.3947326782044243]
inp.current_velocity = [0.1119477497536703, -0.005706738140158095, 0.001348657044096806, 0.1054435897294738, -0.03687579980795278, -0.1371515680506455]
inp.current_acceleration = [-2.943871184141059, 0.1638588832925878, -0.03832918779616604, -3.516112331376623, 1.09743295912854, 4.327530374547349]
inp.target_position = [0, 0, 0, 0, 0, 0]
inp.target_velocity = [0, 0, 0, 0, 0, 0]
inp.target_acceleration = [0, 0, 0, 0, 0, 0]
inp.max_velocity = [6.457718232379019, 5.410520681182422, 7.155849933176751, 9.599310885968812, 9.424777960769379, 17.4532925199433]
inp.max_acceleration = [26.89552377323262, 22.5322006432468, 29.81022362406315, 39.98549316319009, 39.61897402027128, 72.71041663808377]
inp.max_jerk = [224.2224490037115, 187.8497873921497, 248.4650723139127, 333.3055272533571, 330.268654354887, 606.0132228774711]
```
I tried with ```inp.duration_discretization = ruckig::DurationDiscretization::Discrete;``` and it works.

pantor, 2022-03-03:
Hey @fricher, can you check again with the latest commit? There was indeed a bug in the velocity control that should be fixed now.

pantor, 2022-03-04:
This should be fixed with the latest commit. Thanks for reporting!

fricher, 2022-03-04:
Looks good 👍 

Thanks for your help !

closed 2022-03-04 by fricher